chipsalliance/verible #958: Compile Verible on Windows

mikhailbond, 2021-09-28:
Hi all! 
I saw that there were added support to compile Verible on WIndows. But I don't understand what should I do exactly to achieve this.
Could you help me, please, like step by step guide, how to build and run Verible on Windows?

hzeller, 2021-09-28:
A good starting point for what to do is the Continuous Integration script which runs the compile and test.
https://github.com/chipsalliance/verible/blob/master/.github/workflows/verible-ci.yml#L270-L285

Mostly the important part is to use clang (llvm) to compile as there were issues with msvc (@corco knows the details).

But once you've installed clang and bazel, the compile should work as [described in the README](https://github.com/chipsalliance/verible#build). The `bazel install` will not work AFAIK on windows, so you have to manually copy the resulting executables from `bazel-bin/verilog/tools/*/verible-verilog-{syntax,lint,format}`

lf-, 2021-11-03:
I would really appreciate if binaries from compiling Verible on Windows are added to the release artefacts.

hzeller, 2021-11-04:
I have no experience in what a good distribution format is for Windows; accepting pull requests that adds something that is useful for Windows systems.

hzeller, 2021-11-08:
@corco you mentioned in #307 that you'd like to look into extracting windows artifacts from the CI. Are you still up for it ?

corco, 2021-11-11:
Yes, I have been heavily side-tracked on this task ;) I will take another look at it next week