chipsalliance/verible #959: linter: Nonexistent config file doesn't throw an error

mole99, 2021-09-29:
**Describe the bug**

A nonexistent config file doesn't throw an error or display any kind of message.

To reproduce run:

```
./verible-verilog-lint --rules_config="linterrules-nonexistent.txt" my_design.sv
```

where "linterrules-nonexistent.txt" doesn't exist.

**Actual vs. expected behavior**

An error message should be displayed stating that the configuration file could not be found, possibly halting execution.

Currently no message is printed and linting continuous with the standard rule set.

tgorochowik, 2021-09-29:
Thanks for the report!

It the messages are there, but warnings are not displayed by default, for example:
```
$ GLOG_logtostderr=1 verible-verilog-lint --rules_config docdasda test.sv
W20210929 14:56:56.286309 54514 verilog_linter_configuration.cc:318] docdasda: Unable to read rules configuration file NOT_FOUND: docdasda:No such file or directory
```

works and displays meaningful messages (probably even more that any user may need).

I agree however that it makes sense to always display such issues.

@hzeller should we raise the severity of selected messages to error? If not - we should consider printing warnings by default.

hzeller, 2021-09-29:
Raising severity to default print warnings is probably good, but we also should just bubble up this problem up to main() and have it exit with a non-zero exit code.

commit be4eff52 (2021-09-29): Bubble up issues reading the configuration file up to main.

Fixes #959
closed 2021-09-29 by hzeller