chipsalliance/verible #963: test fail when I use sudo bazel test -c opt //...

Banana-Two, 2021-09-30:
//common/util:file_util_test                                             FAILED in 0.1s
  /home/banana/.cache/bazel/_bazel_root/ef4da2825d7ad34a78c6af14a6b3ef53/execroot/com_google_verible/bazel-out/k8-opt/testlogs/common/util/file_util_test/test.log

Executed 1 out of 551 tests: 550 tests pass and 1 fails locally.
There were tests whose specified size is too big. Use the --test_verbose_timeout_warnings command line option to see which ones these areINFO: Build completed, 1 test FAILED, 2 total actions

Banana-Two, 2021-09-30:
my system is ubuntu18.04,and my bazel version is 4.2.1.
I have run the three commands as follow:
sudo bazel build -c opt //... 
sudo bazel run -c opt :install -- -s /usr/local/bin
sudo bazel test -c opt //...
Only the last command will make some errors.
This is log.
exec ${PAGER:-/usr/bin/less} "$0" || exit 1
Executing tests from //common/util:file_util_test
-----------------------------------------------------------------------------
Running main() from gmock_main.cc
[==========] Running 15 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 15 tests from FileUtil
[ RUN      ] FileUtil.Basename
[       OK ] FileUtil.Basename (0 ms)
[ RUN      ] FileUtil.Dirname
[       OK ] FileUtil.Dirname (0 ms)
[ RUN      ] FileUtil.Stem
[       OK ] FileUtil.Stem (0 ms)
[ RUN      ] FileUtil.JoinPath
[       OK ] FileUtil.JoinPath (0 ms)
[ RUN      ] FileUtil.CreateDir
[       OK ] FileUtil.CreateDir (0 ms)
[ RUN      ] FileUtil.StatusErrorReporting
common/util/file_util_test.cc:134: Failure
Value of: status.ok()
  Actual: true
Expected: false
Stack trace:
  0x7fad27638f40: testing::internal::HandleExceptionsInMethodIfSupported<>()
  0x7fad27639158: testing::Test::Run()
  0x7fad27639465: testing::TestInfo::Run()
  0x7fad27639ce9: testing::TestSuite::Run()
... Google Test internal frames ...

common/util/file_util_test.cc:135: Failure
Expected equality of these values:
  status.code()
    Which is: OK
  absl::StatusCode::kPermissionDenied
    Which is: PERMISSION_DENIED
OK
Stack trace:
  0x7fad27638f40: testing::internal::HandleExceptionsInMethodIfSupported<>()
  0x7fad27639158: testing::Test::Run()
  0x7fad27639465: testing::TestInfo::Run()
  0x7fad27639ce9: testing::TestSuite::Run()
... Google Test internal frames ...

[  FAILED  ] FileUtil.StatusErrorReporting (0 ms)
[ RUN      ] FileUtil.ScopedTestFile
[       OK ] FileUtil.ScopedTestFile (0 ms)
[ RUN      ] FileUtil.ScopedTestFileMove
[       OK ] FileUtil.ScopedTestFileMove (0 ms)
[ RUN      ] FileUtil.ScopedTestFileEmplace
[       OK ] FileUtil.ScopedTestFileEmplace (0 ms)
[ RUN      ] FileUtil.FileExistsDirectoryErrorMessage
[       OK ] FileUtil.FileExistsDirectoryErrorMessage (0 ms)
[ RUN      ] FileUtil.UpwardFileSearchTest
[       OK ] FileUtil.UpwardFileSearchTest (0 ms)
[ RUN      ] FileUtil.ReadEmptyDirectory
[       OK ] FileUtil.ReadEmptyDirectory (0 ms)
[ RUN      ] FileUtil.ReadNonexistentDirectory
[       OK ] FileUtil.ReadNonexistentDirectory (0 ms)
[ RUN      ] FileUtil.ListNotADirectory
[       OK ] FileUtil.ListNotADirectory (0 ms)
[ RUN      ] FileUtil.ReadDirectory
[       OK ] FileUtil.ReadDirectory (0 ms)
[----------] 15 tests from FileUtil (2 ms total)

[----------] Global test environment tear-down
[==========] 15 tests from 1 test suite ran. (2 ms total)
[  PASSED  ] 14 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] FileUtil.StatusErrorReporting

 1 FAILED TEST

Banana-Two, 2021-09-30:
my gcc version is 9.4.0

hzeller, 2021-09-30:
This is odd.
What this test is testing is if you attempt to read a file that whose permissions don't allow you to read, that an attempt to read will return a permission denied problem.

The way it does that is to deliberately set the permissions to no permissions, and then attempts to read it.

However, in your case, this test apparently does not see the problem. Maybe setting the limited permissions didn't work ?

Does your home directory that contains the bazel-cache directory maybe has different permission features ?
I can only imagine that on a limited filesystem (such as FAT32) or a network filesystem with insufficient permission settings could be a problem here.

Anyway, in #965 I've refined the test to output more information. Could you do a `git pull` and run the tests again to see if there are interesting things we can see ?

hzeller, 2021-10-01:
@Banana-Two can you run the test with the latest repository and report what it outputs on that failure ?

Banana-Two, 2021-10-02:
> @Banana-Two can you run the test with the latest repository and report what it outputs on that failure ?
I did the following as you said:
cd /home/banana/Downloads/program/verible
git pull https://github.com.cnpmjs.org/chipsalliance/verible.git  
sudo bazel build -c opt //...
sudo bazel run -c opt :install -- -s /usr/local/bin
sudo bazel test -c opt //...
And the same error occured again. 
exec ${PAGER:-/usr/bin/less} "$0" || exit 1
Executing tests from //common/util:file_util_test
-----------------------------------------------------------------------------
Running main() from gmock_main.cc
[==========] Running 15 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 15 tests from FileUtil
[ RUN      ] FileUtil.Basename
[       OK ] FileUtil.Basename (0 ms)
[ RUN      ] FileUtil.Dirname
[       OK ] FileUtil.Dirname (0 ms)
[ RUN      ] FileUtil.Stem
[       OK ] FileUtil.Stem (0 ms)
[ RUN      ] FileUtil.JoinPath
[       OK ] FileUtil.JoinPath (0 ms)
[ RUN      ] FileUtil.CreateDir
[       OK ] FileUtil.CreateDir (0 ms)
[ RUN      ] FileUtil.StatusErrorReporting
common/util/file_util_test.cc:141: Failure
Value of: status.ok()
  Actual: true
Expected: false
Expected permission denied for /home/banana/.cache/bazel/_bazel_root/ef4da2825d7ad34a78c6af14a6b3ef53/sandbox/linux-sandbox/368/execroot/com_google_verible/_tmp/48497f1aa3313bbf82a01afd87baf74f/test-err
Stack trace:
  0x7f0fee48ef40: testing::internal::HandleExceptionsInMethodIfSupported<>()
  0x7f0fee48f158: testing::Test::Run()
  0x7f0fee48f465: testing::TestInfo::Run()
  0x7f0fee48fce9: testing::TestSuite::Run()
... Google Test internal frames ...

common/util/file_util_test.cc:142: Failure
Expected equality of these values:
  status.code()
    Which is: OK
  absl::StatusCode::kPermissionDenied
    Which is: PERMISSION_DENIED
OK
Stack trace:
  0x7f0fee48ef40: testing::internal::HandleExceptionsInMethodIfSupported<>()
  0x7f0fee48f158: testing::Test::Run()
  0x7f0fee48f465: testing::TestInfo::Run()
  0x7f0fee48fce9: testing::TestSuite::Run()
... Google Test internal frames ...

common/util/file_util_test.cc:143: Failure
Value of: content.empty()
  Actual: false
Expected: true
'foo'
Stack trace:
  0x7f0fee48ef40: testing::internal::HandleExceptionsInMethodIfSupported<>()
  0x7f0fee48f158: testing::Test::Run()
  0x7f0fee48f465: testing::TestInfo::Run()
  0x7f0fee48fce9: testing::TestSuite::Run()
... Google Test internal frames ...

[  FAILED  ] FileUtil.StatusErrorReporting (0 ms)
[ RUN      ] FileUtil.ScopedTestFile
[       OK ] FileUtil.ScopedTestFile (0 ms)
[ RUN      ] FileUtil.ScopedTestFileMove
[       OK ] FileUtil.ScopedTestFileMove (0 ms)
[ RUN      ] FileUtil.ScopedTestFileEmplace
[... 153 lines cut ...]
[       OK ] FileUtil.ReadNonexistentDirectory (0 ms)
[ RUN      ] FileUtil.ListNotADirectory
[       OK ] FileUtil.ListNotADirectory (0 ms)
[ RUN      ] FileUtil.ReadDirectory
[       OK ] FileUtil.ReadDirectory (0 ms)
[----------] 15 tests from FileUtil (3 ms total)

[----------] Global test environment tear-down
[==========] 15 tests from 1 test suite ran. (3 ms total)
[  PASSED  ] 14 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] FileUtil.StatusErrorReporting

 1 FAILED TEST

Last, this my /home/banana/.cache/bazel director.
total 16
drwxr-xr-x  4 banana banana 4096 Oct  2 19:46 .
drwxr-xr-x 20 banana banana 4096 Oct  2 09:07 ..
drwxr-xr-x  5 banana banana 4096 Oct  2 19:28 _bazel_banana
drwxr-xr-x  5 root   root   4096 Oct  2 19:46 _bazel_root
This is my /home/banana/.cache
drwxr-xr-x 20 banana banana  4096 Oct  2 09:07 .
drwxr-xr-x 49 banana banana  4096 Oct  2 20:07 ..
drwxr-xr-x  4 banana banana  4096 Oct  2 19:46 bazel

Banana-Two, 2021-10-02:
> @Banana-Two can you run the test with the latest repository and report what it outputs on that failure ?
Besides , I found a thing that the director sandbox doesn't exist. But bazel test -c opt //... didn't report this error while sudo bazel test -c opt //... did.

#banana@Hua in ~/.cache/bazel/_bazel_banana/6ff07553cb0a36d91538ce3fd3e67e9e [20:13:27]
$ pwd                                
/home/banana/.cache/bazel/_bazel_banana/6ff07553cb0a36d91538ce3fd3e67e9e
#banana@Hua in ~/.cache/bazel/_bazel_banana/6ff07553cb0a36d91538ce3fd3e67e9e [20:19:01]
$ ls
action_cache   command.profile.gz  external  java.log.hua.banana.log.java.20211002-192821.18028  README
bazel-workers  DO_NOT_BUILD_HERE   install   javalog.properties                                  server
command.log    execroot            java.log  lock

#banana@Hua in ~/.cache/bazel/_bazel_root/98a4d29cf4bd98a1b687a9955c6a7c98 [20:19:52]
$ pwd
/home/banana/.cache/bazel/_bazel_root/98a4d29cf4bd98a1b687a9955c6a7c98
#banana@Hua in ~/.cache/bazel/_bazel_root/98a4d29cf4bd98a1b687a9955c6a7c98 [20:19:56]
$ ls
action_cache   command.profile.gz  external  java.log.hua.root.log.java.20211002-194654.24418  README
bazel-workers  DO_NOT_BUILD_HERE   install   javalog.properties                                server
command.log    execroot            java.log  lock

hzeller, 2021-10-04:
ah, you're running the test as root `sudo bazel test -c opt //...` which will make the filesystem be lenient in this test.

You never need to run as sudo when compiling verible, in particular you shouldn't as it will make some files in your ~/.cache/bazel owned by root.

Can you start over and first clean your cache directory `sudo rm -rf ~/.cache/bazel`

.. And then run all the commands _not_ as root
```
bazel test -c opt //...
bazel build -c opt //...
bazel run -c opt :install -- -s /usr/local/bin
```

The `:install` rule will ask for the sudo permissions if needed.

closed 2021-10-04 by hzeller

Banana-Two, 2021-10-04:
> ah, you're running the test as root `sudo bazel test -c opt //...` which will make the filesystem be lenient in this test.
> 
> You never need to run as sudo when compiling verible, in particular you shouldn't as it will make some files in your ~/.cache/bazel owned by root.
> 
> Can you start over and first clean your cache directory `sudo rm -rf ~/.cache/bazel`
> 
> .. And then run all the commands _not_ as root
> 
> ```
> bazel test -c opt //...
> bazel build -c opt //...
> bazel run -c opt :install -- -s /usr/local/bin
> ```
> 
> The `:install` rule will ask for the sudo permissions if needed.
> 
> Also see [Install section](https://github.com/chipsalliance/verible#installation) in the README which also mentions that the install process should not be run with sudo.

Sorry, I missed it. Now I have solved the problem as you said. Thanks very much.

PR #970 (2021-10-04): Skip permission test if tests are run as root.
With root permissions, even a non-readable file will be readable.
So if for whatever reason the bazel test is run with root permission,
skip the test.

Fixes #963
commit ec99dcd3 (2021-10-04): Skip permission test if tests are run as root.

With root permissions, even a non-readable file will be readable.
So if for whatever reason the bazel test is run with root permission,
skip the test.

Fixes #963