chipsalliance/verible #1003: CHECK null failure in project tool

hzeller, 2021-10-17:
To reproduce

```
bazel build verilog/tools/project:verible-verilog-project
git clone https://github.com/SymbiFlow/uvm /tmp/uvm
bazel-bin/verilog/tools/project/verible-verilog-project symbol-table-defs --file_list_root=/ --file_list_path <(/tmp/uvm/src/base/uvm_run_test_callback.svh)
```
Output is 
```
F20211017 10:22:03.390353 1917551 symbol_table.cc:1133] Check failed: ::verible::DieIfNull("verilog/analysis/symbol_table.cc", 1133, "ref.components", (ref.components))->Children().size() == 1 (0 vs. 1) 
*** Check failure stack trace: ***
    @     0x5619632aa934  google::LogMessage::Fail()
    @     0x5619632aa87e  google::LogMessage::SendToLog()
    @     0x5619632aa195  google::LogMessage::Flush()
    @     0x5619632adaf6  google::LogMessageFatal::~LogMessageFatal()
    @     0x5619631d697f  verilog::SymbolTable::Builder::LookupOrInjectOutOfLineDefinition()
    @     0x5619631d6ec0  verilog::SymbolTable::Builder::DescendThroughOutOfLineDefinition()
    @     0x5619631d70ac  verilog::SymbolTable::Builder::HandleQualifiedId()
    @     0x5619631d2421  verilog::SymbolTable::Builder::Visit()
...
```

fangism, 2022-03-04:
Reduced test case:

```systemverilog
virtual class uvm_run_test_callback extends uvm_callback;
  extern function new( string name="uvm_run_test_callback");
endclass : uvm_run_test_callback

function uvm_run_test_callback::new( string name="uvm_run_test_callback");
  super.new( name );
endfunction
```

I'm pretty sure I have no code to handle this yet.

fangism, 2022-03-04:
This crashes (same signature):

```systemverilog
virtual class C extends B;
  extern function new;
endclass : C

function C::new();
  super.new();
endfunction
```

but this finishes (changing `new` to `f`):

```systemverilog
virtual class C extends B;
  extern function f;
endclass : C

function C::f();
  super.f();
endfunction
```

commit ffba3629 (2022-03-07): Use 'new' as a constructor identifier.

Handle out-of-line constructor definition case.
Logic is similar to other out-of-line method definitions.

```systemverilog
class C;
  extern function new;  // constructor prototype
endclass
function C::new();  // out-of-line constructor definition
endfunction
```

Fixes #1003
closed 2022-03-07 by hzeller