chipsalliance/verible #1014: Contribute smoke-tests for verible to https://github.com/hdl/smoke-tests

mithro, 2021-10-17:
The https://github.com/hdl organization is trying to create a repository of smoke tests for tools in the EDA (both FPGA and ASIC) ecosystem. The aim is to have tests which can be run after packaging to confirm that the tool still works as expected.

It would be great to have the recent (or soon to be done?) work on improving Verible's smoke-test suite also benefit the packaging efforts.

mithro, 2021-10-17:
Thoughts?

umarcor, 2021-12-07:
The main idea of [hdl/smoke-tests](https://github.com/hdl/smoke-tests) is to be a subset of the CI in each of the projects. 

We assume that projects have their own robust and extensive CI in order to catch errors as close as possible to the source.
That's where the plumbing for improving the tools and finding missing features should be done.
Up to our capacity, we try to collaborate with projects upstream in order to improve their CI.

Conversely, hdl/smoke-tests is meant for packagers of the tools (on system package managers, Conda, containers, homebrew, etc.) to ensure that they did not screw up the build. They need to trigger the minimal set of features to guarantee that some binary is not broken, or some shared library/asset/dependency is not missing.

Ideally, we would be able to do something such as `curl -fsSL https://github.com/chipsalliance/verible/blob/master/.github/bin/smoke-test.sh | sh - -- --minimal`.
In order to make it possible, the `smoke-test.sh` script in this repository would need to support the `--minimal` option implying:

- It is assumed that Verible and any required runtime dependencies are already built and available in the PATH.
- All the assets downloaded by `smoke-test.sh` are kept in some local directory, and it does not install anything on the system.

On a different abstraction layer, in the future we should add "integration tests" to hdl/smoke-tests. Those would be for testing Verible combined with other tools in the ecosystem. An example of an integration test would be the combination of GHDL + Yosys + Surelog.

@hzeller, is it possible to make `smoke-test.sh` usable outside or do you think a `minimal-test.sh` could be added to this repo?