chipsalliance/verible #1035: Two+ consecutive EOL comments in kDPIImportItem cause Check fail.

mglb, 2021-10-20:
**Test case**

```systemverilog
import "DPI-C" context function void foo(
  input bit first,
  // c3
  // c3+
  input bit second
);
```

Include any options or configuration used.

**Actual output**

```
F20211020 12:53:57.903033 134771 align.h:426] Check failed: leading_tokens.empty() || first_tree_token_it == ftokens.end() || first_tree_token_it->before.break_decision != SpacingOptions::MustWrap
*** Check failure stack trace: ***
    @     0x5556a1346f42  google::LogMessage::Fail()
    @     0x5556a1346e7d  google::LogMessage::SendToLog()
    @     0x5556a1346797  google::LogMessage::Flush()
    @     0x5556a134a14a  google::LogMessageFatal::~LogMessageFatal()
    @     0x5556a1205da1  verible::ScanPartitionForAlignmentCells_WithNonTreeTokens<>()
    @     0x5556a11ffa8a  _ZZN7verible29AlignmentCellScannerGeneratorIN7verilog9formatter34PortDeclarationColumnSchemaScannerEEESt8functionIFNS_10VectorTreeINS_19ColumnPositionEntryEEERKNS5_INS_13UnwrappedLineEEEEES4_IFT_vEERKS4_IFvNS_24container_iterator_rangeIN9__gnu_cxx17__normal_iteratorIPKNS_14PreFormatTokenESt6vectorISK_SaISK_EEEEEESR_PS7_EEENKUlSB_E_clESB_
    @     0x5556a120c4e3  _ZNSt17_Function_handlerIFN7verible10VectorTreeINS0_19ColumnPositionEntryEEERKNS1_INS0_13UnwrappedLineEEEEZNS0_29AlignmentCellScannerGeneratorIN7verilog9formatter34PortDeclarationColumnSchemaScannerEEESt8functionIS8_ESD_IFT_vEERKSD_IFvNS0_24container_iterator_rangeIN9__gnu_cxx17__normal_iteratorIPKNS0_14PreFormatTokenESt6vectorISL_SaISL_EEEEEESS_PS3_EEEUlS7_E_E9_M_invokeERKSt9_Any_dataS7_
    @     0x5556a125220d  std::function<>::operator()()
    @     0x5556a12394fb  verible::AlignablePartitionGroup::CalculateAlignmentSpacings()
    @     0x5556a123b899  verible::AlignablePartitionGroup::Align()
    @     0x5556a11f0018  verilog::formatter::TabularAlignTokenPartitions()
    @     0x5556a11df091  _ZZN7verilog9formatter9Formatter6FormatERKNS0_16ExecutionControlEENKUlRN7verible10VectorTreeINS5_13UnwrappedLineEEEE_clES9_
    @     0x5556a11dff45  _ZNSt17_Function_handlerIFvRN7verible10VectorTreeINS0_13UnwrappedLineEEEEZN7verilog9formatter9Formatter6FormatERKNS7_16ExecutionControlEEUlS4_E_E9_M_invokeERKSt9_Any_dataS4_
    @     0x5556a11e4193  std::function<>::operator()()
    @     0x5556a11e24a7  verible::VectorTree<>::ApplyPreOrder()
    @     0x5556a11e2511  verible::VectorTree<>::ApplyPreOrder()
    @     0x5556a11e2511  verible::VectorTree<>::ApplyPreOrder()
    @     0x5556a11e0e09  verible::TreeUnwrapper::ApplyPreOrder()
    @     0x5556a11df403  verilog::formatter::Formatter::Format()
    @     0x5556a11dd0ca  verilog::formatter::FormatVerilog()
    @     0x5556a11bcea1  formatOneFile()
    @     0x5556a11bd72f  main
    @     0x7ff7349270b3  __libc_start_main
    @     0x5556a11bc5de  _start
    @              (nil)  (unknown)
```

**Expected or suggested output**

```systemverilog
import "DPI-C" context function void foo(
import "DPI-C" context function void foo(
  input bit first,
  // c3
  // c3+
  input bit second
);
```

**Why that happens**

This is a check for detecting wrong (unsupported) partitioning: https://github.com/chipsalliance/verible/blob/9f630fbdee2ba9754d4c902665131d70edaf77cc/common/formatting/align.h#L416
```
// Breaking following condition leads to e.g. concatenation of EOL comment
// and code into a single line. To fix situation that lead to this, flatten
// token partitions that contain EOL comment subpartition just before a
// subpartition that starts with the same token as Origin(). Example of a
// partition that needs flattening:
//
//   { (>>[...], (origin: "input bit second"))
//     { (>>[// comment] }
//     { (>>[input bit second], (origin: "input")) }
//   }
```

msierszulski, 2022-12-30:
Hi @fangism @mglb I think this issue can be closed. Currently, the test performed on the master, gives a good result:
test case:
```systemverilog
import "DPI-C" context function void foo(
  input bit first,
  // c3
  // c3+
  input bit second
);
```
result:
```systemverilog
Import "DPI-C" context function void foo(
  input bit first,
  // c3
  // c3+
  input bit second
);
```
I can add a test for this case if needed.

mglb, 2023-01-02:
Please add the test and close, thanks.

PR #1580 (2023-01-06): Add test for two EOL comments in KDPIImportItem
When I was looking for another issue to fix, I came across #1035, which turned out to already work and format the case in a good way. In that case, I add the test to this case, and the issue related to it can be closed.

Fixes #1035.
closed 2023-01-06 by hzeller