chipsalliance/verible #1072: Obfuscator does not discard randomized identifiers that are SystemVerilog keywords.

mglb, 2021-10-30:
Accident catched in the CI when testing a change not related to obfuscator:
```
INFO: From Testing //verilog/tools/obfuscator:obfuscate_test:
==================== Test output for //verilog/tools/obfuscator:obfuscate_test:
(...)
### Testing obfuscate with --save_map
(...)
Test obfuscate --decode mode
Run obfuscator.  Save substitutions.
output is not equivalent: Mismatched token enums.  got: (SymbolIdentifier) (#293: "baz") vs. ("end") (#333: "end")
First mismatched token [14]: (SymbolIdentifier) (#293: "baz") vs. ("end") (#333: "end")

ORIGINAL:
  module foo_bar;
    foo bar(.baz(baz),
      .clk(clk));
  endmodule

ENCODED:
  module dBdVmuM;
    TCY GaI(.end(end),
      .MQj(MQj));
  endmodule

*** Please file a bug. ***
Expected exit code 0, but got 1
```

Obfuscator should discard randomized names that are SystemVerilog keywords.

hzeller, 2021-11-11:
Another example of a randmly failing test in which 'bar' was encoded as `abs`, which made it look like a function call:

```
  module mBRmEcd;
    B6N abs(.nbO(nbO),
      .NdD(NdD));
  endmodule
```

hzeller, 2021-11-11:
And this resulted in something looking like the `pow` function:
```
module foo_bar(input clk);
   UQm WU8(.pow(pow),
     .clk(clk));
endmodule
```

PR #1239 (2022-03-02): Obfuscator: Make sure to not accidentally generating keywords for identifier
Since we can't easily ask the lexer for a definitive list of keywords
that are not SymbolIdentififier's, we simply use it as an oracle while
generating a random name.
    
Fixes #1072
commit e64e36da (2022-03-02): Obfuscator: Make sure to not accidentally generating keywords for identifier

Since we can't easily ask the lexer for a definitive list of keywords
that are not SymbolIdentififier's, we simply use it as an oracle while
generating a random name.

Fixes #1072
closed 2022-03-02 by hzeller