chipsalliance/verible #1074: Pre-processor macro with range is valid.

DaveMcEwan, 2021-11-01:
**Describe the bug**

Short summary.

**To Reproduce**

```systemverilog
`define LSb(n) [n-1:0]
module testcase_ppMacro;
localparam int A = 123;
wire b = A`LSb(5);
wire c = A[5-1:0];
endmodule
```

Include any options used.

**Actual behavior:**
```
./verible-v0.0-1687-g92890a21/bin/verible-verilog-syntax ~/testcase_ppMacro.sv
/home/dm/testcase_ppMacro.sv:4:11: syntax error, rejected "`LSb" (syntax-error).
```

**Expected behavior**

A clear and concise description of what you expected to happen. Citations to the
[SystemVerilog-2017 Standard (LRM)](https://ieeexplore.ieee.org/document/8299595)
would help.
Wires `b` and `c` should be equivalent.
6.20.2, page 121, "Bit-selects and part-selects of parameters that are of integral types shall be allowed (see 6.11.1)." 

This may be related to #951

hzeller, 2021-11-02:
Yes, this is indeed a known limitation currently, as there is limited preprocessing support currently (see also #812).
It is on the TODO list, and we get to it as resources permit (also, help welcome :) ).

CC @fangism

fangism, 2021-11-02:
Right, we have **no preprocessor** today, but instead preprocessing constructs are explicitly woven into the grammar with many limitations.  Depending on the application, we could mitigate the problem with improved error-recovery and ignoring problematic excerpts of code, or require an actual preprocessor.

fangism, 2021-11-02:
There is also an unfinished feature of psuedo-preprocessing using definitions local to a single file.
https://cs.opensource.google/verible/verible/+/master:verilog/preprocessor/verilog_preprocess.h

The idea is to expand preprocessor macros for the sake of parsing using only local definitions, but allow operation on that file using either an expanded view (for syntax checking and linting) or unexpanded view (for formatting).

DaveMcEwan, 2021-11-03:
Doing this properly will require more than just pseudo-preprocessing macros defined in the same file.
I've found this issue in (https://github.com/DaveMcEwan/dmpvl/blob/master/hdl/cdcFifo.sv) which does this:
```
`include "misc.svh"
...
// Use of `LSb causing issue.
```
and misc.svh looks has the macro definition.

A "quick'n'dirty" approach could be to ignore any syntax checks between a macro and a semicolon.
The "proper" approach might be to perform syntax checks on an expanded AST (as opposed to style checks on the source AST). That would also need flags to set incdirs. The benefit would be that all the wacky things people use pp macros for would be supported.

I'm no C++ expert but if it's possible to develop on CentOS7 with GCC 4.8 (yes, sadly I'm restricted) then I can look at doing it.

raynnersc, 2026-09-03:
Hi, is this issue still active, or has it already been resolved?
 Just checking if there are any updates or workarounds available. Thanks!