chipsalliance/verible #1091: Formatter do not consider inline lint waivers

y-kim, 2021-11-08:
**Test case**

```systemverilog
module test;
  always_comb begin
    // semantically illegal, but it is used to make lint error
    a <= b; b <= c; // verilog_lint: waive always-comb-blocking
  end
endmodule
```

**Actual output**

```systemverilog

module test;
  always_comb begin
    // semantically illegal, but it is used to make lint error
    a <= b;
    b <= c; // verilog_lint: waive always-comb-blocking
  end
endmodule
```

Include any possible diagnostic messages from the formatter.

**Expected or suggested output**

```systemverilog
module test;
  always_comb begin
    // semantically illegal, but it is used to make lint error
    a <= b; // verilog_lint: waive always-comb-blocking
    b <= c; // verilog_lint: waive always-comb-blocking
  end
endmodule
```

```systemverilog
module test;
  always_comb begin
    // semantically illegal, but it is used to make lint error
    a <= b; b <= c; // verilog_lint: waive always-comb-blocking
  end
endmodule
```

Expand a lint waive into multiple lines or do not split a line when it has an inline lint waiver.

y-kim, 2021-11-08:
Here is anoter example with `--try_wrap_long_lines` option

*Original code*
```systemverilog
module test;
  logic a2345678901234567890123456789012345678901234567890;
  logic b2345678901234567890123456789012345678901234567890;

  always_comb begin
    // semantically illegal. but intended.
    a2345678901234567890123456789012345678901234567890 <= b2345678901234567890123456789012345678901234567890; // verilog_lint: waive always-comb-blocking
  end
endmodule
```

*Actual output*
```systemverilog
module test;
  logic a2345678901234567890123456789012345678901234567890;
  logic b2345678901234567890123456789012345678901234567890;

  always_comb begin
    // semantically illegal. but intended.
    a2345678901234567890123456789012345678901234567890
        <= b2345678901234567890123456789012345678901234567890
        ; // verilog_lint: waive always-comb-blocking
  end
endmodule
```

*Expected or suggested output*
```systemverilog
module test;
  logic a2345678901234567890123456789012345678901234567890;
  logic b2345678901234567890123456789012345678901234567890;

  always_comb begin
    // semantically illegal. but intended.
    a2345678901234567890123456789012345678901234567890
        <= b2345678901234567890123456789012345678901234567890 // verilog_lint: waive always-comb-blocking
        ;
  end
endmodule
```

```systemverilog
module test;
  logic a2345678901234567890123456789012345678901234567890;
  logic b2345678901234567890123456789012345678901234567890;

  always_comb begin
    // semantically illegal. but intended.
    a2345678901234567890123456789012345678901234567890
        <= b2345678901234567890123456789012345678901234567890; // verilog_lint: waive always-comb-blocking
  end
endmodule
```

hzeller, 2021-11-08:
The linter and formatter are somewhat indpendent and the formatting examples here look like essentially working-as-intended. The problem is the linting waiver that stretches over multiple lines that verible does not deal with correctly yet.

What the formatter attempts to do is to fit things according to the formatting rules - lint waivers are only considered comments at that point with no special meaning to the formatter.
In particular it can't make guesses that migh be unintended, say in the case of `a <= b; b <= c; ` was it meant to add the linting to both statements in that line or was that accidental in the formatting before ? It is safer for the formatter to re-organize the code according to the formatting rules, and then if that exposes new places where you'd like to add linter rules, you have to add them there manually once.

Now in the other case, there are two parts to it: 
 1. how the formatter processes the lines (I think it looks correct) and
 2. the lint-waiver applies the waive to the broken-up line (should improve):

  1) The formatting looks weird but it is the only reasonable thing in this case. Here the long `<=` line is formatted with the semicolon in a new line. But that is due to the very long lines here as

```SystemVerilog
    a2345678901234567890123456789012345678901234567890 <= b2345678901234567890123456789012345678901234567890; // verilog_lint: waive always-comb-blocking
```
exceeds 100 characeters, and even one break will exceed 100 characters
```SystemVerilog
    a2345678901234567890123456789012345678901234567890 <= 
      b2345678901234567890123456789012345678901234567890; // verilog_lint: waive always-comb-blocking
```

.. So the solution is to break at two places while keeping the integrity of the whole statement which includes carrying over the semicolon.

```SystemVerilog
    a2345678901234567890123456789012345678901234567890 <=
      b2345678901234567890123456789012345678901234567890
      ; // verilog_lint: waive always-comb-blocking
```

(Note, the formatter will never change the code, i.e. it will never move a comment to another place (like in front of the semicolon), so this is the only semantically correct choice if we want to keep the comment technically at the end of the line.
The suggested examples you gave will not obey the line length or the no-semantic change paradigms.

 2) The next should be applied to the _semantic_ line not the physical line where the waiver is given; I think this should indeed be improved. After the formatting, the semantic line is split in three physical lines, but they should all be considered as the 'semantic line before the semicolon'. This currently also prevents to just have a comment in front of the line.

y-kim, 2021-11-09:
I agree with your comment for the first case. It is better to re-write code so a lint waiver to be applied to only one statement (or one item). I was just afraid a case that backward propagation is prohibited, but it will be minor case neglectable.

Actually my problem was the second one. So I hope improvement of the linter.

Thanks.