chipsalliance/verible #1103: [signal-name-style]  rule seg faults when 'sample' is used as a net name.

craigldunn, 2021-11-18:
signal-name-style - Does not like the word 'sample' as a variable name. 

Code compiles on 2x EDA simulators. 

Seem to only affect this rule, disabling rule works around the issue.

The following produces a seg fault  on CentOs 7
use tag v0.0-1752-g8836b293 also seen on v0.0-1633
```echo "module ABC; reg sample; endmodule" |  verible-v0.0-1752-g8836b293/bin/verible-verilog-lint --ruleset=none --rules=+signal-name-style -```
```
*** Aborted at 1637257250 (unix time) try "date -d @1637257250" if you are using GNU date ***
PC: @                0x0 (unknown)
*** SIGSEGV (@0x18) received by PID 11538 (TID 0x7fecabcfe740) from PID 24; stack trace: ***
    @           0x507ea7 google::(anonymous namespace)::FailureSignalHandler()
    @     0x7fecab3eb630 (unknown)
    @           0x4d3c75 verilog::analysis::SignalNameStyleRule::HandleSymbol()
    @           0x46554c verible::SyntaxTreeLinter::Visit()
    @           0x4e6941 verible::TreeContextVisitor::Visit()
    @           0x4e6941 verible::TreeContextVisitor::Visit()
    @           0x4e6941 verible::TreeContextVisitor::Visit()
    @           0x46526c verible::SyntaxTreeLinter::Lint()
    @           0x437eed verilog::VerilogLintTextStructure()
    @           0x438066 verilog::LintOneFile()
    @           0x41d381 main
    @     0x7fecaae1a555 __libc_start_main
    @           0x425bfe (unknown)
    @                0x0 (unknown)
Segmentation fault
```

fangism, 2021-11-23:
`sample` is listed as a keyword.

https://cs.opensource.google/verible/verible/+/master:verilog/parser/verilog.lex;drc=dd5409bebbee407e48fe25a01e5a42e57c336532;l=468

In the LRM(2017), Annex A.2.1 describes `coverage_event`:

```
coverage_event ::=
  clocking_event
| with function sample ( [ tf_port_list ] )
| @@( block_event_expression )
```

where both `function` and `sample` are keywords.

fangism, 2021-11-23:
However, we should not crash under any circumstances.

craigldunn, 2021-11-23:
Thanks for taking a look 

However I disagree with the assertion that 'sample' is keyword. 'sample' is a built in method of a covergroup and the example you sight it the way you override it.

See
LRM(2017)
19.8.1 Overriding the built-in sample method

There are of course reserved keyword for net identifiers 

>5.6.2 Keywords
>Keywords are predefined nonescaped identifiers that are used to define the language constructs. A
>SystemVerilog keyword preceded by an escape character is not interpreted as a keyword.
>All keywords are defined in lowercase only. Annex B gives a list of all defined keywords.

However sample does not appear in Annex B