chipsalliance/verible #1108: How to pass a file list to style linter

rahulraveendran15-coder, 2021-11-22:
Hi,

How can I pass the file list to the linter tool command?
I have so many files to lint and passing each and every thing is command line is difficult. Is there an option to pass -f <filelist>

fangism, 2021-11-23:
There's no filelist option at the moment.

However, you could use a shell tool like `xargs` like so:

```
cat filelist.f | xargs -n 1 verible-verilog-lint
```

or

```
find . -name '*.sv' | xargs -n 1 verible-verilog-lint
```

rahulraveendran15-coder, 2021-11-25:
So, if this is the case I cannot use relative paths in my file list, since  the verible-verilog-lint depends on the path I am running

rahulraveendran15-coder, 2021-11-26:
Also, one more issue I found out is piping the output in terminal to a file is not working with verible-verilog-lint command. How can I solve this?
verible-verilog-lint test.sv | tee test.log