chipsalliance/verible #1111: Formatting a file that is included in another file

SeanMcLoughlin, 2021-11-22:
**Test case**

```systemverilog
// my_module.sv
module my_module;
  `include "my_included_file.sv"
endmodule
```

```systemverilog
// my_included_file.sv
generate
  genvar i;
  for (i = 0; i < 4; i++) begin
    // do stuff
  end
endgenerate
```

I want to format `my_included_file.sv`, but Verible rejects its syntax, presumably because it thinks the generate block isn't wrapped in a module. 

It's a bit of an annoyance as a user because I have a lot of code that is like this, but I'm guessing that the fix for this on Verible's side is pretty complicated. Or maybe this is beyond the scope of what Verible provides. So I'll ask two questions:

1. My current workaround to get my files formatted is to comment out these generate blocks, run the formatter, un-comment them, and manually format them. Is there a simpler workaround?
2. What is the end-goal for Verible for issues like this? Your README specifies:
> It was born out of a need to parse un-preprocessed source files, which is suitable for single-file applications like style-linting and formatting. In doing so, **it can be adapted to parse preprocessed source files, which is what real compilers and toolchains require.**

So will this project expand to include something like a language server in the future? Or is the goal just to have Verible act as the back-end for someone else's implementation of a language server?

fangism, 2021-11-23:
Verible needs a hint that what it's looking at is internal to a module:
https://github.com/chipsalliance/verible/tree/master/verilog/analysis#alternative-parsing-modes

Try inserting near the top:

```
// verilog_syntax: parse-as-module-body
```

SeanMcLoughlin, 2021-11-23:
This works. Thanks!

closed 2021-12-08 by fangism