chipsalliance/verible #1112: Format fails when a `ifdef is in a else if chain

Rki009, 2021-11-28:
**Test case**
module bug_001 (
	input clk,
	output x
);

	reg [7:0] junk;
	always @(posedge clk) begin
		if (do_this) junk <= 8'h01;
		else if (do_that) junk <= 8'h02;
`ifdef ADD_THIS_CODE
		else if (do_extra) junk <= 8'h03;
`endif
		else junk <= 8'h05;
	end

endmodule

**Actual output**

+ ../bin/verible-verilog-format bug_001.v
bug_001.v: bug_001.v:11:3: syntax error, rejected "else" (syntax-error).
bug_001.v:13:3: syntax error, rejected "else" (syntax-error).

**version**
v0.0-1758-g036cd0d3
Commit	2021-11-26 04:47:08 -0800
Built	2021-11-26T13:16:03Z

fangism, 2021-12-08:
The above test case isn't rendering properly, could you re-paste it inside triple-ticks like this:

```
code
```

I'm not sure if this is due to poor code-rendering, but did you mean `` `else `` and not ``else``?

This is a known weakness in the parser implementation.  Verible doesn't have a preprocessor yet, so parsing with preprocessing directives is very limited.  It can work when the directives are placed around whole statement syntactic boundaries, and the clauses themselves contain whole statements.

davekeeshan, 2023-07-12:
verible version: [76cc3fad49a10c0bb69c519e71ea6965e6dd121c](https://github.com/chipsalliance/verible/commit/76cc3fad49a10c0bb69c519e71ea6965e6dd121c)

I see this issue in code where the io is controlled by an `ifdef`, usually seen when added debug interfaces:
```verilog
module core (
    output aresetn
    , input s_aclk
    , output uart_rxd_out
    , input uart_txd_in
    `ifdef D_TEST_AXI
    , input [3:0] t_axi_awid
    , input [31:0] t_axi_awaddr
    , input [7:0] t_axi_awlen
    , input [2:0] t_axi_awsize
    , input [1:0] t_axi_awburst
    , input [0:0] t_axi_awlock
    , input [3:0] t_axi_awcache
    , input [2:0] t_axi_awprot
    , input [3:0] t_axi_awqos
    , input [0:0] t_axi_awvalid
    , output [0:0] t_axi_awready
    , input [31:0] t_axi_wdata
    , input [3:0] t_axi_wstrb
    , input [0:0] t_axi_wlast
    , input [0:0] t_axi_wvalid
    , output [0:0] t_axi_wready
    , output [3:0] t_axi_bid
    , output [1:0] t_axi_bresp
    , output [0:0] t_axi_bvalid
    , input [0:0] t_axi_bready
    , input [3:0] t_axi_arid
    , input [31:0] t_axi_araddr
    , input [7:0] t_axi_arlen
    , input [2:0] t_axi_arsize
    , input [1:0] t_axi_arburst
    , input [0:0] t_axi_arlock
    , input [3:0] t_axi_arcache
    , input [2:0] t_axi_arprot
    , input [3:0] t_axi_arqos
    , input [0:0] t_axi_arvalid
    , output [0:0] t_axi_arready
    , output [3:0] t_axi_rid
    , output [31:0] t_axi_rdata
    , output [1:0] t_axi_rresp
    , output [0:0] t_axi_rlast
    , output [0:0] t_axi_rvalid
    , input [0:0] t_axi_rready
    `endif    
);
```
This is what failure looks like
```
../src/common/core.sv: ../src/common/core.sv:67:5: syntax error at token ","
../src/common/core.sv:122:5-10: syntax error at token "assign"
../src/common/core.sv:123:5-10: syntax error at token "assign"
../src/common/core.sv:125:5-10: syntax error at token "genvar"
../src/common/core.sv:207:5-12: syntax error at token "generate"
../src/common/core.sv:208:23: syntax error at token "<"
../src/common/core.sv:208:47: syntax error at token ")"
```
What is annoying is that I tried to wrapper the offending code with the disabling-formatting tags, these were ignored, and it fails the same way as above.

```verilog
    // verilog_format: off 
    `ifdef D_TEST_AXI
    // verilog_format: on
```

I have seen this code style in loads of companies.

kbrunham-intel, 2026-09-22:
@hzeller This issue still does not have a reduced, copy-pasteable `.sv` reproducer that we can run with `verible-verilog-format`. Could the original reporter (or anyone hitting this) please attach a minimal snippet? Without that it is hard to confirm whether the bug is still present on v0.0-4294-gc1d8f5e8 or to add a regression test.