chipsalliance/verible #1113: Formatter fails if there's a line comment following a delay

urish, 2021-12-03:
**Test case**

```systemverilog
module test ();
  reg x;
  initial begin
    #20
    // test
    x = 1;
  end
endmodule
```

**Actual output**

```systemverilog
$ verible-verilog-format src/test.v
module test ();
  reg x;
  initial begin
    #20
    // test
    x = 1;

  end
endmodulesrc/test.v: Formatted output is lexically different from the input.    Please file a bug.  Details:
") vs. ("// end of line comment") (#748: "// test")nt") (#748: "// test
; problematic formatter output is
module test ();
  reg x;
  initial begin
    x = 1;/ test

  end
endmodule
<<EOF>>
```

fangism, 2021-12-08:
It looks like the `#20` was dropped accidentally, and the `// test` comment was ruined.

hzeller, 2022-01-15:
Still broken with the current version:
What I get with the current version is that it merges the end of line comment with the `x = 1;` and thus hiding it.
And comment is placed after the `#20` timing - which is also not proper.

It looks like some tokens are missing preceeding newlines (can you have a look @mglb ?) 

```SystemVerilog
module test ();
  reg x;
  initial begin
    #20
    // test
    x = 1;
  end
endmodule
test-files/format-break.sv: Error lex/parsing-ing formatted output.  Please file a bug.
First error: token: "end" at 5:3-5; problematic formatter output is
module test ();
  reg x;
  initial begin
    #20  // test    x = 1;
  end
endmodule
```

mglb, 2022-01-17:
I'll look into this. Seems like a "blind" meging of trailing ";" (which is in a single partition with `x = 1` tokens) with preceding partition without checking whether it is possible.

yatian-liu, 2022-01-17:
> 

Can confirm this, just experienced the same bug on my computer.

PR #1568 (2023-01-10): Fix comment following a delay
This PR fixes the following case:
```systemverilog
module test ();
  reg x;
  initial begin
    #20 //test
    x = 1;
    x = 2;
  end
endmodule
```

Fixes #1113.
closed 2023-01-10 by hzeller