chipsalliance/verible #1114: [Feature req] Lint Rule for width mismatch

tushar-uwaterloo, 2021-12-05:
**Summary**
There should be a lint rule that reports when the LHS and RHS signals are of different width. There is already a `undersized-binary-literal` and `truncated-numeric-literal` lint rules exist, but they only work with constant values.

**Test cases and examples**
logic [3:0] a;
logic [1:0] b;

assign a = b; //this should generate a width mismatch warning, since RHS signal 'b' is smaller in length than LHS signal 'a'.
assign b = a; //this should also generate a width mismatch warning, since LHS signal 'a' is bigger in length than RHS signal 'b'.

tushar-uwaterloo, 2021-12-05:
Another example is when a module is instantiated within a module and there is a width mismatch:

```
module top
(       input [5:0] a,
        output [4:0] b,
);

under u_inst
(  .a  (a),   //u_inst.a is 2 bits wide  
   .b  (b)    //u_inst.b is 6 bits wide
);
endmodule : top

module under
(       input [1:0] a,
        output [5:0] b,
);

assign b = {4'hF,a};

endmodule : under
```

fangism, 2025-01-23:
What you're asking here requires more type evaluation than what is currently available in Verible, as widths may contain expressions, macro values, parameters, etc.
But that said, if you are looking for something best-effort that can handle obviously constant widths (and conservatively ignore other cases), we'd need to start with adding more type checking.
The place one might start this is from the symbol table code, which currently has basic symbol binding information and unprocessed type information bound to declarations -- it has enough information to tell you what text (and where) types came from, but it doesn't do any evaluation yet.