chipsalliance/verible #1118: something controlled by macro is not supported.

CannedGrape, 2021-12-10:
**Test case**

```systemverilog
// Input to the formatter, preferably a reduced test case.
`ifdef A
module_name2 (
`else
module_name2 (
`endif
u_module (
...
```

**Actual output**

Syntax Error.

fangism, 2021-12-14:
Verible doesn't have a real preprocessor (yet), so parsing examples like the above won't work.  
There is limited support where preprocessing directives fall on who syntactic construct boundaries, and can yield a syntax-tree with preprocessing directives in it.

Gallium70, 2022-02-23:
Another example that Verible can't parse:

```verilog
`define DEF_A

module top;
  wire a, b, c, d;
  assign d = { a,
`ifdef DEF_A
    b,
`endif
    c
  };
endmodule
```