chipsalliance/verible #1119: [Feature Req] instantiation alignment

CannedGrape, 2021-12-10:
**Test case**

```systemverilog
// Input to the formatter, preferably a reduced test case.
module_a_top(
    .signal_a (wire_signal_a),
    .long_signal_b (wire_long_signal_b)
)
```

**Actual output**

```systemverilog
module_a_top(
    .signal_a      (wire_signal_a),
    .long_signal_b (wire_long_signal_b)
)
```
**Expected or suggested output**

```systemverilog
// This result would look better from the formatter.
module_a_top(
    .signal_a      (wire_signal_a     ),
    .long_signal_b (wire_long_signal_b)
)
```

fangism, 2021-12-14:
It sounds like you want a style option where the close parentheses are aligned in their own column, rather than flushing left.
(I've seen examples like this before.)

CannedGrape, 2021-12-14:
yes~

birdybro, 2021-12-28:
I second this feature request. In the MiSTer Project we use this syntax as an example:

```sv
module EEPROM_X24CXX
  (//Replace type_24C01 with EEPROM size and command size (Test State y/n and address bytes)?
    input             type_24C01,          // 24C01 is 128 bytes/no Test State?, 24C02 is 256 bytes/Test State
    input             type_24C04,
    input             type_24C08,
    input             type_24C16,
    input             type_24C65,
    input             type_X24C01,
    input             type_X24C02,
    input      [ 3:0] page_mask,           // Typically 0x3, 0x7, or 0xF; semi-correlated with size
    input             no_test_state,       // No control word (ID check/upper address bits); Usually only for X24C01
    input             address_write_only,  // Some models only set address in write mode (24C0xA?)
    input             clk,
    input             ce,
    input             reset,
    input             SCL,                 // Serial Clock
    input             SDA_in,              // Serial Data (same pin as below, split for convenience)
    output reg        SDA_out,             // Serial Data (same pin as above, split for convenience)
    input      [ 2:0] E_id,                // Chip Enable ID
    input             WC_n,                // ~Write Control
    input      [ 7:0] data_from_ram,       // Data read from RAM
    output     [ 7:0] data_to_ram,         // Data written to RAM
    output     [ 7:0] ram_addr,            // RAM Address
    output reg        ram_read,            // RAM read
    output reg        ram_write,           // RAM write
    input             ram_done             // RAM access done
);
```
It would be nice if this was used. :)

suzizecat, 2022-02-24:
I erm... third (?) this ! 
It is useful to have a 
```systemverilog
module_a_top(
    .signal_a      (wire_signal_a     ),
    .long_signal_b (wire_long_signal_b)
)
```
output since it allows an easy square selection of all signals without catching parenthesis.