chipsalliance/verible #1120: [Feature Req] port alignment

CannedGrape, 2021-12-10:
**Test case**

```systemverilog
// Input to the formatter, preferably a reduced test case.
module module_name (
    input signal_a,
    input [`WIDTH-1:0]signal_b,
    output reg signal_c,
    output wire signal_d,
    output long_signal_e
);

```

**Actual output**

```systemverilog
module module_name (
    input                    signal_a,
    input       [`WIDTH-1:0] signal_b,
    output reg               signal_c,
    output wire              signal_d,
    output                   long_signal_e
);
```

**Expected or suggested output**

```systemverilog
// This result would look better from the formatter.
module module_name (
    input                    signal_a     ,
    input       [`WIDTH-1:0] signal_b      ,
    output reg               signal_c     ,
    output wire              signal_d     ,
    output                   long_signal_e
);
```

snsokolov, 2021-12-10:
Is the feature basically asking for aligning commas?

CannedGrape, 2021-12-14:
yes, this alignment can be convenient for integrator.

CannedGrape, 2021-12-17:
That's OK~
I usually do this alignment in my own style.  I copy port list and replace "," with "()," when module instantiation, and then other operations in Column-mode of vim/VSCode.

suzizecat, 2022-02-24:
Bumping a bit on this one. This is the way SublimeText systemverilog does it and is quite neat when you have to add comments to describe the ports. 

Also, It looks nicer IMHO to get this nice and square block rather than having commas "all over the place"