chipsalliance/verible #1122: Build error on Fedora 35

jagjordi, 2021-12-13:
I am getting the following error while trying to build verible on fedora 35:

```
➜  verible git:(master) bazel build -c opt //...
Starting local Bazel server and connecting to it...
INFO: Analyzed 1864 targets (111 packages loaded, 5362 targets configured).
INFO: Found 1864 targets...
bazel: Entering directory `/home/jordi/.cache/bazel/_bazel_jordi/13967d7c87a057fae689ac53148986f9/execroot/com_google_verible/'
ERROR: /home/jordi/.cache/bazel/_bazel_jordi/13967d7c87a057fae689ac53148986f9/external/m4_v1.4.18/gnulib/BUILD.bazel:192:11: Compiling gnulib/lib/c-stack.c [for host] failed: (Exit 1): gcc failed: error executing command /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG -ffunction-sections ... (remaining 25 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
In file included from /usr/include/signal.h:328,
                 from external/m4_v1.4.18/gnulib/lib/c-stack.c:49:
external/m4_v1.4.18/gnulib/lib/c-stack.c:55:26: error: missing binary operator before token "("
   55 | #elif HAVE_LIBSIGSEGV && SIGSTKSZ < 16384
      |                          ^~~~~~~~
external/m4_v1.4.18/gnulib/lib/c-stack.c:139:8: error: variably modified 'buffer' at file scope
  139 |   char buffer[SIGSTKSZ];
      |        ^~~~~~
bazel: Leaving directory `/home/jordi/.cache/bazel/_bazel_jordi/13967d7c87a057fae689ac53148986f9/execroot/com_google_verible/'
INFO: Elapsed time: 10.128s, Critical Path: 3.96s
INFO: 42 processes: 13 internal, 29 linux-sandbox.
FAILED: Build did NOT complete successfully
```
My toolset version:
```
➜  ~ g++ --version
g++ (GCC) 11.2.1 20210728 (Red Hat 11.2.1-1)
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

➜  ~ bazel --version
bazel 4.2.1
```

Any idea why this happens?
I was able to build in RHEL8 but not in Fedora35, which is strange.

fangism, 2021-12-14:
We should see if this is reproducible from `m4` source directly (external dependency), and see if they already have some fix.

corco, 2021-12-14:
@fangism I tested version 1.4.18 (current) and 1.4.19 (latest). The issue exist on current and is fixed in latest.

I will see if we can add 1.4.19 to rules_m4

jagjordi, 2021-12-14:
Thank you for your answer, could you instruct on how to change from `current` to `latest` ? I understand that this is related to bazel configuration and not verible. I am not familiar with bazel system. Thank you

hzeller, 2021-12-14:
For Verible, we have #957 to track this. There is no solution yet as upgrading the bazel m4 rules are a bit tricky.
There are solutions being discussed to simply use the native `m4` that is on any Unix system anyway.

Passing on to @tgorochowik to look into details (also cc @QuantamHD who had a first look in the context of https://github.com/hdl/bazel_rules_hdl )

tgorochowik, 2021-12-28:
@jagjordi could you please check if #1132 resolves this for you?

jagjordi, 2021-12-28:
I'm not home right now, I will check in a couple of days

jagjordi, 2021-12-29:
@tgorochowik it seems to work fine! Thank you!

closed 2022-01-12 by hzeller

kiteloopdesign, 2022-03-17:
Facing a similar problem on Fedora 35.

**Building with:**
`bazel build -c opt  --//bazel:use_local_flex_bison //...`

It builds with no errors.

**Installing with:** 
`bazel run -c opt :install --//bazel:use_local_flex_bison //.. -- -s /usr/local/bin`

Doesn't raise an actual ERROR, but some issue related to permissions, see below. Nothing gets installed on `/usr/local/bin`. Also tried with the ~/bin local option and I am getting the same issue

```
Target //:install up-to-date:
  bazel-bin/_install_gen.sh
  bazel-bin/install
INFO: Elapsed time: 165.146s, Critical Path: 28.79s
INFO: 433 processes: 19 internal, 414 linux-sandbox.
INFO: Build completed successfully, 433 total actions
INFO: Build completed successfully, 433 total actions
install: cannot create regular file '//../verible-patch-tool': Permission denied
```

mglb, 2022-03-18:
> **Installing with:**
> `bazel run -c opt :install --//bazel:use_local_flex_bison //.. -- -s /usr/local/bin`

This command doesn't look legit. Try: `bazel run -c opt --//bazel:use_local_flex_bison //:install -- -s /usr/local/bin`

kiteloopdesign, 2022-03-18:
Ah, the fine user-friendly syntax of `bazel`!

Working, thanks!

PR #1132 (2022-01-12): Use host m4/flex/bison on linux
This should allow us to skip the problematic update of the m4 dependency. Other operating systems can keep using the outdated version, Linux should be allowed to use host tools.

Closes #1122 
Closes #957