chipsalliance/verible #1155: can't build in macos using bazel

Zhangyanting-1997, 2022-01-26:
**Describe the bug**

verible/verilog/parser/BUILD:122:11: @platforms//os:windows is not a valid configuration key for //verilog/parser:verilog_y_cc

command line : bazel build //verilog/tools/syntax:verible-verilog-syntax

hzeller, 2022-01-26:
In the CI, the MacOS build succeeds ( https://github.com/chipsalliance/verible/runs/4928395460 )
What is your bazel version ? (output of `bazel --version`)

hzeller, 2022-01-28:
Did you get it to work @Zhangyanting-1997 ?

C1oud555, 2022-03-01:
I can build on macos 12. But the direct install isn't working, I have to manually copy all the binaries to the desired folder.

hzeller, 2023-08-01:
Please provide a patch to the install script. I don't have a Mac so I can't test that, so it is up to someone who has access to such machine.
But looks like there is work-in-progress in that install project https://github.com/google/bazel_rules_install/pull/37