chipsalliance/verible #1165: CentOS8 docker build starting to fail

hzeller, 2022-01-31:
https://github.com/chipsalliance/verible/runs/5013790643

Looks like it can't download from a URL for a repo. This might either be a temporary glitch on that server or a changed URL in general that requires to update the docker to have the URLs it uses point to the right one.

hzeller, 2022-01-31:
Unai, that looks like something you might be interested in.

umarcor, 2022-02-01:
It's failing when executing `yum install`, so it's a problem upstream. Either with the container image or with their package repositories. I tried locally with a freshly pulled `centos:8` container, and I get the same error. Therefore, I believe it's unrelated to verible or the content installed in the CI of this repo.

hzeller, 2022-02-01:
Thanks for investigating!
Right now, any failure in the release builds make the full release fail so they will [go stale](https://github.com/chipsalliance/verible/releases/) (last is from a couple of days ago).

Maybe skipping such failed build but still continuing with the releases might be good. Or mabye we should just disable CentOS8 for now ?

corco, 2022-02-01:
CentOS 8 has reached EOL in [December](https://www.centos.org/centos-linux-eol/), to be replaced by CentOS stream (rolling release).

I would suggest removing CentOS 8 releases.

We can evaluate if we want to support CentOS stream later? I am not sure *who* uses CentOS stream as a development platform for hardware design. From CentOS stream website

"
Continuously delivered distro that tracks just ahead of Red Hat Enterprise Linux (RHEL) development, positioned as a midstream between Fedora Linux and RHEL. For anyone interested in participating and collaborating in the RHEL ecosystem, CentOS Stream is your reliable platform for innovation.
"

umarcor, 2022-02-01:
@corco thanks!

I would propose using Fedora instead of CentOS as the testing platform for "the RHEL family of distros". Fedora maintaines two active releases: https://endoflife.date/fedora.

However, what was the reason for testing on CentOS? Are there known users of Verible which need it?

> Maybe skipping such failed build but still continuing with the releases might be good. Or mabye we should just disable CentOS8 for now ?

I believe disabling centos for now is acceptable.

corco, 2022-02-01:
I do not know the history of that decision, but RHEL 7&8 and SUSE Enterprise Linux are commonly the *only* supported platforms by many EDA tools like Modelsim, which makes supporting that platform natural.

umarcor, 2022-02-01:
> I do not know the history of that decision, but RHEL 7&8 and SUSE Enterprise Linux are commonly the _only_ supported platforms by many EDA tools like Modelsim, which makes supporting that platform natural.

That was correct for some decades, because RHEL was the most (only) known company providing commercial support for a Linux distro. However, it started changing some years ago, due to Ubuntu. Nowadays, most EDA tools do support Ubuntu. See:

- https://www.intel.com/content/www/us/en/support/programmable/support-resources/design-software/os-support.html
- https://support.xilinx.com/s/article/54242?language=en_US

Note https://www.xilinx.com/support/documentation/sw_manuals/xilinx2021_2/ug973-vivado-release-notes-install-license.pdf:

> *This is the last version of Xilinx tools supporting CentOS 8.1, 8.2, and 8.3.*
>
> *This is the final version of Xilinx tools supporting 16.04.5 LTS and 16.04.6 LTS.*

In practice, I found Siemens' and Xilinx's tools to work on either Fedora or Debian.

PR #1167 (2022-02-01): Removed CentOS 8 release as it is EOL
CentOS 8 is EOL and fails to build as a consequence

Closes #1165
closed 2022-02-01 by hzeller