chipsalliance/verible #1168: googletest library just changed content

hzeller, 2022-02-01:
The googletest release [mentioned in the WORKSPACE](https://github.com/chipsalliance/verible/blob/master/WORKSPACE#L17-L23) just changed the content:
https://github.com/google/googletest/archive/refs/tags/release-1.11.0.zip

This used to have sha256sum of `353571c2440176ded91c2de6d6cd88ddd41401d14692ec1f99e35d013feda55a`, now it is `63afcb1690b2139b2d9c7ad1df3de87697530fb1fd6869622cff26c77a3b0cfb`.
This could be benign (e.g. github was changing their zip library and repackaged things), but is worth investigating first before changing our expected hash.

mglb, 2022-02-01:
Recursive diff (`diff -r -u new/googletest-release-1.11.0 fromcache/googletest-release-1.11.0`) of both zips didn't find any differences.

hzeller, 2022-02-01:
Relevant https://twitter.com/tgummerer/status/1488493440103030787