chipsalliance/verible #1180: [Feature Request] Auto-generate waiver file from existing violations

ganoam, 2022-02-04:
Hi everyone,

Thanks for all your valuable work.

In our current project, we are looking to start using verible-lint in order to perform style checks on our project. 
Our current code base however includes a ton of verible lint violations, which we will have to sort through (most are trivally fixable)
To facilitate our effort it would be great to have an option to generate a waiver file waiving existing violations. 
It would help us make sure we don't add new violations and help check and fix existing ones.
I took inspiration for this request from similar functionality featured in the verilator project.

I'm happy to help with the implementation, but I am not familiar with your code base and I would need some pointers as to where to start.

cheers,
Noam

mglb, 2022-02-04:
Hi!

Implementation suggestion:

* Create a new subclass of [`ViolationHandler`](https://github.com/chipsalliance/verible/blob/164e27c8477ab3258b06b2f1742d19d956ad9f2e/common/analysis/violation_handler.h#L28-L43). To be more precise - just straight up copy [`ViolationPrinter`](https://github.com/chipsalliance/verible/blob/164e27c8477ab3258b06b2f1742d19d956ad9f2e/common/analysis/violation_handler.h#L45-L59) implementation.
* In `HandleViolations` iterate over all violations and print them in a format required by waiver files. Similarly to [`ViolationPrinter::HandleViolations`](https://github.com/chipsalliance/verible/blob/164e27c8477ab3258b06b2f1742d19d956ad9f2e/common/analysis/violation_handler.cc#L50-L60). It would be nice to put code for generating waiver text in a function somewhere near (or in?) [lint_waiver.cc](https://github.com/chipsalliance/verible/blob/164e27c8477ab3258b06b2f1742d19d956ad9f2e/common/analysis/lint_waiver.cc) file.
* Add new [`AutofixMode`](https://github.com/chipsalliance/verible/blob/164e27c8477ab3258b06b2f1742d19d956ad9f2e/verilog/tools/lint/verilog_lint.cc#L41-L47) (+ strings in a map and help message below) and create the new violation handler in [a switch](https://github.com/chipsalliance/verible/blob/164e27c8477ab3258b06b2f1742d19d956ad9f2e/verilog/tools/lint/verilog_lint.cc#L168-L189).
* I think it would make sense to use a file name from `--autofix_output_file` as an output for lint rules.

ganoam, 2022-02-04:
Thanks @mglb. I'll look into it.

closed 2022-04-01 by hzeller