chipsalliance/verible #1187: Language Server: implement textDocument/hover

hzeller, 2022-02-11:
The Hover request should return some information that can be useful in the context

Ideas:

 * Variables: type if available (maybe requires symbol table, but could probably be more simply derived; register/wire/parameter/localparam; if possible to determine without going too much in elaboration-weeds: width. Input, output ?)
 * Blocks: when hovering over the `end` block - if this is a labelled block, name of the label
 * generalized to `endmodule`, `endfunction`... name of the corresponding thing.

Some simple things can probably be relatively easily be derived, others might need symbol table.

Saad-Mufti, 2023-12-29:
Is there any plan for adding hover info for literals, say to show what a binary literal is in hex or decimal? Or utility functions (e.g. `$clog2`)/other keywords (`always_ff`), to show parameters + basic info? If not I'd be willing to take a stab at it.

hzeller, 2024-01-05:
There are many ideas that could be implemented, alas there is not enough time.
So yes, please take a stab!

thomas-woehrle, 2024-11-11:
Hi, 

I just wanted to ask which hover functionalities currently exist if any. 

[The language server README](https://github.com/chipsalliance/verible/tree/master/verilog/tools/ls) says: _verible-verilog-ls by default loads and analyses only currently edited files in the editor. To be able to utilize such features as going to definition, going to references, **printing hover info** project-wide, add a verible.filelist file to the project._ 

This indicates that hover info already exists, but it doesn't work for me, while go-to-definition does after setting up verible.filelist, hence my question.

hzeller, 2024-11-11:
There is actually some partial hover implementation, but it is disabled right now.

You have to set the `hoverProvider` value to true and recompile: https://github.com/chipsalliance/verible/blob/master/verilog/tools/ls/verilog-language-server.cc#L90

Reason why it is currently disabled is, that it creates a full symbol table of all files first, while it would be probably better to look in the local file first before doing an expensive symbol table creation.

But, this is a start and can be expanded from there.

ButterSus, 2025-02-04:
Hi,

As a workaround, you can use the `hover.nvim` plugin to implement the default hover behavior. Here's how:
https://gist.github.com/ButterSus/36d47999251030ddf3bbfd187046fe7a

This solution leverages nvim-treesitter and expands the preview until the line containing the definition is fully selected.

hzeller, 2025-06-14:
The flag to enable the experimental hover feature is `--lsp_enable_hover`