chipsalliance/verible #1188: LanguageServer: how to tell language server which are relevant files to resolve symbols from

hzeller, 2022-02-11:
Unlike other languages, often in (System)Verilog the source of declarations or definitions are not quite clear as includes are more rarely used; instead, the project is composed by all the files put together by the build system.

So in order to build a proper symbol table, we need to know all the files involved (similar what is passed to the [project tool](https://github.com/chipsalliance/verible/tree/master/verilog/tools/project) and [kythe extractor](https://github.com/chipsalliance/verible/tree/master/verilog/tools/kythe). In this case however, the tool is running in the background, started by the editor.

Check out how this is 'typically' done. I suspect tools such as `clangd` read the compilation database, but there is no such standard in the Verilog world. The only somewhat agreed-upon 'standard' is a filelist.
Maybe we can require that there is a file-list somewhere in the root directory of the project with a particular name ?
A flag that is passed to the language server at start-up time ? (some editors provide ways to customize the invocation of the LSP)
We can also add a custom JSON RPC request to convey that into the language server (highly specialized. Maybe there is already something in the LSP spec with such intent ?)

nblei, 2022-02-13:
The veridian project (https://github.com/vivekmalneedi/veridian) has a yaml format for this purpose.

gasrodriguez, 2022-06-04:
+1 for a "standard" filelist so the same can be used for other tools, eg verilator. It may use relative paths to the filelist location itself (as it called with -F instead of -f) and allow env var substitution. 

Either passed as argument to the ls binary (with -F) or with a predefined file name is OK (maybe `files.verible_ls`?).

To be completely standard I think it should allow macro definitions with +define, include directories with +incdir, lib directories with -y, and (optionally) lib files extension with -libext.

Btw, would be awesome if the macros defined in the file list are honored by the ls somehow, eg turning off unreached code due to some `ifdef, or ignoring it by find definitions etc.

SeanMcLoughlin, 2022-11-26:
> A flag that is passed to the language server at start-up time?

I think this might be the best approach. [DVT](https://dvteclipse.com/documentation/sv_vscode/) (a licensed SystemVerilog IDE and language server) uses [build configuration files](https://dvteclipse.com/documentation/sv_vscode/Multiple_.build_Files.html) where you can specify filelists for the language server to care about.

hzeller, 2022-12-13:
Assigning to Greg for now as it is close to the work he is currently doing.

ljanyst, 2023-03-10:
Is anyone working on this and has made any progress? If not, I'd like to look into it.