chipsalliance/verible #1205: Multiple Directives Merged Into a Single Line

gkielian, 2022-02-19:
Hi There! Tried using verible-verilog Finding that multiple directives are being pulled together on a single line.

This doesn't seem to happen for most of the directives, but it does for the following examples:

Tested this with older version from last summer 

v0.0-1318-gf6b4485 (from a commit on 2021-08-02)

As well as latest available at time of writing:

v0.0-1921-g348c1f8b (from a more recent commit on 2022-02-18)

**Test case**
For a typical verilog file I may have a couple of directives near the top:

```verilog
`default_nettype none
`timescale 1ns / ps
```

After formatting with `verible-verilog-format --inplace filename.v` (calling from 
inside a neovim session), or from `verible-verilog-format filename.v` from the command line,
the two directives are placed on a single line:

**Actual output**
```verilog
`default_nettype none `timescale 1ns / 1ps
```

**Expected or suggested output**

Ideally this would result in the following, for formatting purposes, as well as practicality -- as the single line format cannot be built:

```verilog
`default_nettype none
`timescale 1ns / 1ps
``` 

Finally, no messages are output from the linter or formatter to indicate something is off.

Thanks for your help!

zephray, 2022-12-14:
Encountering the same issue here... is there any workaround for this?

ljmatkins, 2023-12-20:
I am having the same problem as well.

One workaround is to add an empty comment in between:
```
`timescale 1ns / 1ps
//
`default_nettype none
```

or 
```
`timescale 1ns / 1ps  /* */
`default_nettype none
```

kbrunham-intel, 2026-09-22:
@hzeller I retested the reduced case from this issue with `verible-verilog-format` built from tag `v0.0-4294-gc1d8f5e8`. I could not reproduce the problem (formatter exits 0; output keeps the expected layout / converges). I would suggest closing this issue.