chipsalliance/verible #1217: Aligning port declarations

ariyonaty, 2022-02-24:
**Test case**

```systemverilog
module sync_ram #(
  parameter ADDR_WIDTH = 10, DATA_WIDTH = 8 ) (
  input  logic  clk, we, reset,
  input  logic [ADDR_WIDTH - 1:0] addr,
  output logic [DATA_WIDTH - 1:0] data
);
endmodule
```

Flag options used:
`--formal_parameters_alignment=align`
` --port_declarations_alignment=align`

**Actual output**

```systemverilog
module sync_ram #(
  parameter ADDR_WIDTH = 10,
            DATA_WIDTH = 8
) (
  input  logic                    clk,
  we,
  reset,
  input  logic [ADDR_WIDTH - 1:0] addr,
  output logic [DATA_WIDTH - 1:0] data
);

endmodule
```

**Expected or suggested output**

```systemverilog
module sync_ram #(
    parameter ADDR_WIDTH = 10,
              DATA_WIDTH = 8
) (
    input  logic                    clk,
                                    we,
                                    reset,
    /*OR*/
    input  logic                    clk, we, reset,
    input  logic [ADDR_WIDTH - 1:0] addr,
    output logic [DATA_WIDTH - 1:0] data
);

endmodule
```

Version used:
```
v0.0-1854-g0834c7f8
Commit  2022-01-27 17:26:51 -0800
Built   2022-01-28T01:53:49Z
```

---

I would assume that `--port_declarations_alignment=align` should behave similar to `--formal_parameters_alignment=align`.

tokoro10g, 2022-06-13:
I am having the same issue.
A quick inspection of your test case with `verible-verilog-syntax --printtree` shows the following:

```
...
      Node @5 (tag: kParenGroup) {
        Leaf @0 (#'(' @65-66: "(")
        Node @1 (tag: kPortDeclarationList) {
          Node @0 (tag: kPortDeclaration) {
            Leaf @0 (#"input" @69-74: "input")
            Node @2 (tag: kDataType) {
              Node @1 (tag: kDataTypePrimitive) {
                Leaf @0 (#"logic" @76-81: "logic")
              }
              Node @3 (tag: kPackedDimensions) {
              }
            }
            Node @3 (tag: kUnqualifiedId) {
              Leaf @0 (#SymbolIdentifier @83-86: "clk")
            }
            Node @4 (tag: kUnpackedDimensions) {
            }
          }
          Leaf @1 (#',' @86-87: ",")
          Node @2 (tag: kPort) {
            Node @0 (tag: kPortReference) {
              Node @0 (tag: kUnqualifiedId) {
                Leaf @0 (#SymbolIdentifier @88-90: "we")
              }
            }
          }
          Leaf @3 (#',' @90-91: ",")
          Node @4 (tag: kPort) {
            Node @0 (tag: kPortReference) {
              Node @0 (tag: kUnqualifiedId) {
                Leaf @0 (#SymbolIdentifier @92-97: "reset")
              }
            }
          }
          Leaf @5 (#',' @97-98: ",")
...
```

I guess the nodes `2` and `4` should be treated as `kPortDeclaration` rather than `kPort`.

tokoro10g, 2022-06-14:
I'm not sure if this could be a proper solution, but here is a patch for this issue:
https://github.com/tokoro10g/verible/commit/9a92aaa6feac68a30e91bc9fc230793f1d6dda9b