chipsalliance/verible #1236: Long chains of associative operator expressions result in tree whose processing typically crashes the stack

hzeller, 2022-03-01:
Not so much in hand-written code, but in machine generated code, we often see huge chains of expressions with the same associative operator. These end up in a degenerated parse tree essentially being a linked list but are typically processed recursively and make several of the verible tools very unhappy.

let's generate such a long expression:

`gen-long_expr.sh`
```bash
#!/bin/bash
LEN=1000000
echo -n "module long_expr(input logic in[$LEN]); wire combine = 0 "
for (( i=0; i < $LEN; ++i)); do
   echo -n "^in[$i]"
done
echo ";endmodule"
```

```
./gen-long_expr.sh > long_expr.sv
```

Processing these, the tools suffer from walking the stack or simply segfault due to recursion limits they run into

```
verible-verilog-syntax long_expr.sv
verible-verilog-project symbol-table-defs --file_list_root=. --file_list_path <(echo "long_expr.sv")
verible-verilog-kythe-extractor --file_list_root=. --file_list_path <(echo "long_expr.sv")
verible-verilog-lint long_expr.sv
verible-verilog-format long_expr.sv
```

It might be worthwhile to internally replace the super-deep nested tree with an expression node for the particular operator with a large amount of sibling children that can be processed iteratively.

hzeller, 2022-03-01:
Assigning to David who was looking for a recreational programming task.

fangism, 2022-03-01:
Flattening associative operations in grammar actions as the syntax tree is being built is feasible.

PR #1246 (2022-03-04): Flatten associative binary operations.
This reduces stack depth for long chained expressions.

Fixes #1236
commit 03450267 (2022-03-04): Flatten associative binary operations.

This reduces stack depth for long chained expressions.

Fixes #1236
closed 2022-03-04 by hzeller