chipsalliance/verible #1243: Compiling absl/random/internal/randen_hwaes.cc [for host] failed: (Exit 1): gcc failed: error executing command /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG -ffunction-sections ...

richmaes, 2022-03-03:
While building on Ubuntu ARM, running as a VM on an M1 MacBook, seeing an error that appears relative to Google Abseil.   

Linux purple 5.13.0-27-generic #29~20.04.1-Ubuntu SMP Fri Jan 14 00:35:28 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux

INFO: From Executing genrule //verilog/parser:verilog_lex:
verilog/parser/verilog.lex:299: warning, rule cannot be matched
verilog/parser/verilog.lex:972: warning, rule cannot be matched
ERROR: /home/richmaes/.cache/bazel/_bazel_richmaes/9b7688f518eeffaa5605685004095b90/external/com_google_absl/absl/random/internal/BUILD.bazel:323:11: Compiling absl/random/internal/randen_hwaes.cc [for host] failed: (Exit 1): gcc failed: error executing command /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG -ffunction-sections ... (remaining 38 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox
In file included from external/com_google_absl/absl/random/internal/randen_hwaes.cc:229:
/usr/lib/gcc/aarch64-linux-gnu/9/include/arm_neon.h: In function 'Vector128 {anonymous}::AesRound(const Vector128&, const Vector128&)':
/usr/lib/gcc/aarch64-linux-gnu/9/include/arm_neon.h:12452:1: error: inlining failed in call to always_inline 'uint8x16_t vaesmcq_u8(uint8x16_t)': target specific option mismatch
12452 | vaesmcq_u8 (uint8x16_t data)
      | ^~~~~~~~~~
external/com_google_absl/absl/random/internal/randen_hwaes.cc:255:20: note: called from here
  255 |   return vaesmcq_u8(vaeseq_u8(state, uint8x16_t{})) ^ round_key;
      |          ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from external/com_google_absl/absl/random/internal/randen_hwaes.cc:229:
/usr/lib/gcc/aarch64-linux-gnu/9/include/arm_neon.h:12438:1: error: inlining failed in call to always_inline 'uint8x16_t vaeseq_u8(uint8x16_t, uint8x16_t)': target specific option mismatch
12438 | vaeseq_u8 (uint8x16_t data, uint8x16_t key)
      | ^~~~~~~~~
external/com_google_absl/absl/random/internal/randen_hwaes.cc:255:20: note: called from here
  255 |   return vaesmcq_u8(vaeseq_u8(state, uint8x16_t{})) ^ round_key;
      |          ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
At global scope:
cc1plus: warning: unrecognized command line option '-Wno-pass-failed'
INFO: Elapsed time: 36.288s, Critical Path: 7.80s
INFO: 364 processes: 166 internal, 198 linux-sandbox.
FAILED: Build did NOT complete successfully

hzeller, 2022-03-03:
I've updated absl to the latest release ( https://github.com/chipsalliance/verible/commit/991490ccc1e4332efc6d63caffdb7afaeec10f27 ) that looks like it fixed the issue according to the other issues you mention. Can you try again ?

richmaes, 2022-03-04:
I have confirmed that this fix resolves the compile issue.  Thanks for fixing that so promptly.

closed 2022-03-04 by richmaes