chipsalliance/verible #1247: tree-unwrapper: paren_group not found situation detected

hzeller, 2022-03-03:
**Test case**

```systemverilog
module foo ();
  $display($some_systemfunction);
endmodule
```

Formatting produces a result, but also complains about **formatting of macro call failed: paren_group not found**:

```
 echo 'module foo ();$display($some_systemfunction);endmodule' | bazel-bin/verilog/tools/formatter/verible-verilog-format -
E20220303 15:47:49.190407 1964513 tree_unwrapper.cc:2061] formatting of macro call failed: paren_group not found.
*** Please file a bug. ***
E20220303 15:47:49.192584 1964513 tree_unwrapper.cc:2061] formatting of macro call failed: paren_group not found.
*** Please file a bug. ***
module foo ();
  $display($some_systemfunction);
endmodule

```

hzeller, 2022-03-04:
Possibly because it is allowed to have some [`$systemfunctions` without calling parenthesis](https://github.com/chipsalliance/verible/blob/master/verilog/parser/verilog.y#L7753-L7754).

joennlae, 2022-08-01:
Same error can happen when one has something like this:

```sv
`ifdef VERILATOR
  initial begin
    $display("Latch-based register file not supported for Verilator simulation");
    $fatal; // works with $fatal();
  end
`endif
```

LiveQiu, 2022-09-22:
I have meet the same problem, I try delete $display and $finish, it won't happen.

`verible-verilog-format our.v
E0922 16:44:57.371420   17619 tree_unwrapper.cc:2064] formatting of macro call failed: paren_group not found.
*** Please file a bug. ***
E0922 16:44:57.371513   17619 tree_unwrapper.cc:2064] formatting of macro call failed: paren_group not found.
*** Please file a bug. ***
module our;
  initial begin

    $finish;
  end
endmodule`

LiveQiu, 2022-09-22:
I have meet the same problem, I try delete $display and $finish, it won't happen.

```verilog
module our;
    initial begin
        $display("Hello World");
        $finish;
    end
endmodule
```
This is my error. How can fix it. thanks 

verible-verilog-format our.v
E0922 16:44:57.371420   17619 tree_unwrapper.cc:2064] formatting of macro call failed: paren_group not found.
*** Please file a bug. ***
E0922 16:44:57.371513   17619 tree_unwrapper.cc:2064] formatting of macro call failed: paren_group not found.
*** Please file a bug. ***
module our;
  initial begin
    $display("Hello World");
    $finish;
  end
endmodule

PR #1544 (2022-12-06): Fix calling system functions without parentheses
This commit removes the error message when formatting system functions without parentheses, because they are formatted well after all.

Fixes #1280,
Fixes #1223,
Fixes #1247,
Fixes #1392.
commit d591c89c (2022-12-06): Fix calling system functions without parentheses

This commit removes the error message when formatting
system functions without parentheses, because they
are formatted well after all.

Fixes #1280,
Fixes #1223,
Fixes #1247,
Fixes #1392.
closed 2022-12-06 by hzeller