chipsalliance/verible #1280: Error message is printed when formatting assertion with $fatal

virs-ac, 2022-03-22:
**Test case**

The following valid system verilog code results in the formatter printing and error message and suggesting I file a report.  If I remove the `$fatal` line of code the formatter runs without error.

```bash
v0.0-2008-gbbd498f0
Commit  2022-03-13
Built   2022-03-14T07:43:22Z
```

```systemverilog
module decimate_channel(
    input logic aclk,
    input logic tvalid,
    input logic tready
) ;

    assert property (@(posedge aclk) tvalid |-> tready)
    else begin
        $error("Error Messages");
        $fatal;
    end

endmodule    
```

I'm using the default options, see below for the command that I used

**Actual output**

```bash
$ verible-verilog-format bug2.sv
E20220322 14:24:23.078382 2187218 tree_unwrapper.cc:2061] formatting of macro call failed: paren_group not found.
*** Please file a bug. ***
E20220322 14:24:23.082552 2187218 tree_unwrapper.cc:2061] formatting of macro call failed: paren_group not found.
*** Please file a bug. ***
module decimate_channel (
    input logic aclk,
    input logic tvalid,
    input logic tready
);

  assert property (@(posedge aclk) tvalid |-> tready)
  else begin
    $error("Error Messages");
    $fatal;
  end

endmodule
```

**Expected or suggested output**
I would expect the output to be error free

hankhsu1996, 2022-05-27:
Same issue here. Based on IEEE Standard for SystemVerilog (IEEE Std 1800-2017), some system tasks like `$finish`, `$fatal` allow users not to add the parenthesis pair behind. 

Currently, `$stop`, `$finish`, `$exit`, `$time`, `$stime`, `$realtime`, `$printtimescale`, `$timeformat`, `$fatal`, `$error`, `$warning`, `$info` and more system tasks without parenthesis are parsing incorrectly.

prokie, 2022-11-23:
Any progress on this?

PR #1544 (2022-12-06): Fix calling system functions without parentheses
This commit removes the error message when formatting system functions without parentheses, because they are formatted well after all.

Fixes #1280,
Fixes #1223,
Fixes #1247,
Fixes #1392.
commit d591c89c (2022-12-06): Fix calling system functions without parentheses

This commit removes the error message when formatting
system functions without parentheses, because they
are formatted well after all.

Fixes #1280,
Fixes #1223,
Fixes #1247,
Fixes #1392.
closed 2022-12-06 by hzeller