chipsalliance/verible #1301: Add builds for latest CentOS version (CentOS 8 stream and CentOS 9 steam)

mithro, 2022-03-29:
Currently we are only building for CentOS 7 as CentOS 8 reached End Of Life (EOL) on December 31st, 2021 (see https://www.centos.org/centos-linux-eol/ and https://github.com/chipsalliance/verible/pull/1167).

CentOS 8 was replaced with CentOS 8 stream. There is also CentOS 9 stream. See https://www.centos.org/centos-stream/

aronowski, 2022-09-14:
@mithro @umarcor There already are free rebuilds of RHEL 8 and 9 like [EuroLinux](https://en.euro-linux.com/eurolinux/what-is/). Would it be possible to use it as a base?

umarcor, 2022-09-14:
@aronowski there are container images available: https://hub.docker.com/_/rockylinux and https://hub.docker.com/u/eurolinux. Hence, it's technically possible. See https://github.com/chipsalliance/verible/blob/master/releasing/docker-run.sh.

aronowski, 2022-09-14:
@umarcor I'd still be opting for EuroLinux - the company has a way faster release cycle than Rocky. We can compare them on Distrowatch:
- https://distrowatch.com/table.php?distribution=eurolinux

Feature | 9.0 | 8.6
-- | -- | --
Release Date | 2022-06-14 | 2022-05-11

- https://distrowatch.com/table.php?distribution=rocky

Feature | 9.0 | 8.6
-- | -- | --
Release Date | 2022-07-14 | 2022-05-16

I'd be happy to help implement it but I had trouble with building the project - several Bazel-related issues.

gogito, 2022-09-20:
I would like to request building binaries for centOS 6/rhel 6 since a lot of semiconductor companies I work for are still using these and it would be very helpful.