chipsalliance/verible #1302: Add binary releases for all architectures that CentOS supports

mithro, 2022-03-29:
CentOS supports multiple architectures but we only seem to be building for x86. We should release binaries for all the architectures that CentOS supports.

From https://www.centos.org/centos-stream/

| Architecture |
| -- |
| x86_64 | 
| ARM64 (aarch64) | 
| IBM Power (ppc64le) | 
| IBM Z (s390x) |

mithro, 2022-03-29:
@umarcor - Can we use the BuildX / QEMU thingy to make this happen on GitHub Actions?

umarcor, 2022-03-29:
There are no official foreign architecture containers for CentOS: https://github.com/docker-library/official-images#architectures-other-than-amd64. It can be done with Fedora 34, Fedora 35, Fedora 36, Fedora 37, Debian Buster, Debian Bullseye, Ubuntu 16.04, Ubuntu 18.04, Ubuntu 20.04, Ubuntu 21.10, Ubuntu 22.04, etc. Nonetheless, IMHO that'd be better done in copr, salsa, hdl/conda*, hdl/containers, or any other location focused on packaging.

cbalint13, 2022-04-20:
@mithro , @umarcor 

For {epel8, fc35, fc36, fc37} x {aarch64, ppc64le, x86_64} weekly in [HDL repo](https://copr.fedorainfracloud.org/coprs/rezso/HDL) .

Latest build: https://copr.fedorainfracloud.org/coprs/rezso/HDL/build/4293471

* ```centos9``` when the EPEL part of 9 will be settled (many packages are missing ATM). 
* ```s390x``` can be enabled too but it is unattractive (ppc64le is fine IMO).