chipsalliance/verible #1303: Add container build / releases

mithro, 2022-03-29:
It would be good to publish docker container builds / releases for Verible with the stand alone binaries.

mithro, 2022-03-29:
@umarcor - Any thoughts on the best way to do this?

umarcor, 2022-03-29:
I'd suggest contributing to https://github.com/hdl/containers/tree/main/debian-bullseye.
https://hdl.github.io/containers/dev/Contributing.html#single-file-example or https://hdl.github.io/containers/dev/Contributing.html#example-with-hdlc-script can be used as a reference. See also https://hdl.github.io/containers/dev/Contributing.html#step-by-step-checklist.

The target architectures to build the containers for are defined in https://github.com/hdl/containers/blob/main/utils/pyHDLC/config.yml#L128-L143. I would start with `SysDebianAmd64` and then change to `SysDebianDefaultArchSet`.