chipsalliance/verible #1320: [lsp] Spawning language server with cmd: `verible-verilog-ls` failed with error message: EIO: i/o error

eunchan, 2022-04-21:
I've setup verible-verilog-ls in vscode first. "Go to definition" did not work on vscode.

So, I tried on Neovim, by adding:

```
if has('nvim')
lua << EOF
require'lspconfig'.verible.setup {}
EOF
endif
```

Then I see the error message when open `.sv` file in nvim.

```
Spawning language server with cmd: `verible-verilog-ls` failed with error message: EIO: i/o error
```

Anyone let me know how I can get more useful debug information to resolve?

eunchan, 2022-04-21:
OS: CentOS 7

```
vim --version
NVIM v0.6.0
Build type: Debug
LuaJIT 2.1.0-beta3
```

msfschaffner, 2022-04-22:
@hzeller

eunchan, 2022-04-22:
Ah nevermind. It seems to be OK now in NeoVIM. I will check vscode too.

msfschaffner, 2022-04-22:
Was there anything specific that you had to do to get it running?

msfschaffner, 2022-04-22:
FYI: I am running this on gLinux with Sublime build 4126 and it works fine for me.

hzeller, 2022-04-22:
Are there things we should add to the README  ? Maybe it didn't find it in your path the first time ?

eunchan, 2022-04-22:
OK looks like some environment variable prevents the lsp from running. After removing all environment scripts in my bash, it works.

vscode, however, still struggles to do "go to definition". It sais "No definitions found for ''"

hzeller, 2022-04-22:
Expected - "go to definition" is still in the testing phase and not yet upstreamed. Stay tuned.

huanghx1990, 2022-08-20:
hello guys， it is have any update for this issue? I also could not use go to defination in vscode

IEncinas10, 2023-04-12:
Should this be closed? The language server works correctly in my setup

```
NVIM v0.7.2
Build type: Release
LuaJIT 2.1.0-beta3
```