chipsalliance/verible #1327: [WIP][Discussion] Standalone preprocessor design

karimtera, 2022-04-30:
# SV Standalone Preprocessor Design

## Introduction

The goal of this thread is to discuss the design of a standalone System-Verilog preprocessor.
The preprocessor should fully support the directives described in 2018 SV-LRM, and provide APIs that can be used to build a standalone tool (token in / token out basis).

This proposal is a continuation on the already built pseudo-preprocessing.
Also, refer to #1318 to track the current progress.

## Functions (APIs)
### ScanStream (_DONE_)
#### Parameters
- Input: constant reference to a token stream (full SV source code).
- Return: `VerilogPreprocessData`
#### Algorithm
1. Calls `MakeConstIteratorStreamer`, to create an iterator generator.
2. Generates iterators using generator, calls `HandleTokenIterator` for each token iterator.
3. Return a move from the `VerilogPreprocessData`.

### HandleTokenIterator (_DONE_)
#### Parameters
- Input: constant iterator to a token stream view.
- Input: steam iterator generator (is needed for cases when more than one token is consumed). 
#### Algorithm
1. This function is a mainly a case switch, that calls the needed function for each directive.
2. Calls `HandleDefine`, `HandleIf`, `HandleMacroIdentifier`, and so on when needed.
3. Forwards the token to the preprocessed token stream view.
4. Returns status.

### HandleMacroIdentifier (_IN PROGRESS_)
#### Parameters
- Input: Token iterator (pointing to a macro call e.g. `MACRO).
- Output: is appended to preprocessor memory 
#### Algorithm
1. Calls `ConsumeMacroCall`.
2. Calls `ParseMacroCall`.
3. Find the macro definition that matches the macro being called [^1].
4. Calls `ExpandMacro`.
6. Add the last vector of tokens in the preprocessor memory to the preprocessor output stream.
7. Return status.

### ExpandMacro (_IN PROGRESS_)
#### Parameters
- Input: `MacroCall` (populated by `ParseMacroCall`)
- Input: `MacroDefinition`
- Output: is appended to preprocessor memory 
#### Algorithm
1. Begin to re-lex the macro definition body
8. Forward lexed tokens to the output sequence
9. A lexed token should is directly forwarded to the output sequence, unless the lexed token is:
   - A formal parameter in the `MacroDefinition` then substitute with `ExpandParameter` [^2], then forward the last vector of tokens in the preprocessor memory to output sequence.
    - Another macro identifier, then call `HandleMacroIdentifier`, then forward the last vector of tokens in the preprocessor memory to output sequence.
    - A preprocessor directive, `HandleTokenIterator` should be called, then forward the last vector of tokens in the preprocessor memory to output sequence.
10. Append the output sequence to the preprocessor backup memory (which is a vector of token sequences).

Return status.

### ParseMacroCall (_IN PROGRESS_)
#### Parameters
- Input: `TokenStreamView` (populated by `ConsumeMacroCall`).
- Output: `MacroCall`.
#### Algorithm
1. Populate `MacroCall` by parsing the `TokenStreamView`.
4. `MacroCall` should contain:
   - Macro name.
   - Flag if it has parameters.
   - Parameters as a vector of tokens .
11. Return status.

### ConsumeMacroCall (_IN PROGRESS_)
#### Parameters
- Input: stream iterator generator by ref (if we need to consume more than one token).
- Input: `TokenStreamView` iterator (pointing to the macro name).
- Output: a `TokenStreamView` pointer.
#### Algorithm
1. Extract macro name (directly pointed to by iterator)
2. Find its definition
3. Determine is it callable or not
12. Populates a `TokenStreamView`
    - Until ‘)’ if callable
    - Until just the end of the macro name if not callable
13. Return status

[^1]: Finding the macro definition is done 2 times in HandleMacroIdentifier, that should be optimized.
[^2]: ExpandParameter should work exactly as ExpandMacro, but maybe with diff parameters?

karimtera, 2022-05-05:
Question: I am thinking instead of adding a handling method for each directive inside `ExpandMacro`, It would be a better idea to just use `ScanStream`? Because that would be the end result anyway.

For example, right now only these tokens are correctly interpreted inside `ExpandMacro`:
- Actual parameters being replaced.
- Other macro calls (with or without parameters).
- Other tokens that are not preprocessor directives.

So, If we find some way to use `ScanStream` on the macro body itself, that would solve all the limitations we have right now.

tgorochowik, 2022-05-13:
CC @hzeller @mglb

hzeller, 2022-06-24:
Also cc @fangism