chipsalliance/verible #1335: verible-verilog-ls cannot attach in neovim

fimwest, 2022-05-12:
My OS is windows. I downloaded the lastest verible binaries and add path of verible in environment varibles. 
My neovim verision is 0.7.0. I also installed nvim-lspconfig and enabled the verible config.
However verible-verilog-ls cannot attach in neovim. 
The lsp.log shows .../vim/lsp/rpc.lua:420	"rpc"	"verible-verilog-ls"	"stderr"	"Verible Alpha Language Server 0.0 alpha\r\n".

r2com, 2022-06-02:
any success?

hzeller, 2022-06-02:
I don't have access to a Windows machine, but I'll try to reproduce this on a Linux machine (so far I have only tested  the language server successfullywith emacs, vscode, kate and sublime on Linux). So, work in progress.
Sorry, currently immersed in some other project, so a little slow to respond.

closed 2023-03-20 by fimwest