chipsalliance/verible #1342: Problem with verible-verilog-ls attaching in neovim

r2com, 2022-06-02:
I have neovim setup with few other language LSP's working, I installed the verible through nvim-lsp-install, installation looks OK, executable files added where needed.

The server was configured properly after lspconfig.

However, opening basic .sv file says that no LSP is present, the :LspInfo shows output as in attached pic below:
[https://i.imgur.com/95iCRSw.jpg](url)

any ideas why it is not working? I am just opening basic .sv file inside folder where there is nothing else.

r2com, 2022-06-02:
[image]

this is the output

paller, 2022-06-15:
I believe it's because it fails to detect your project's root directory as it only looks for git repositories. Run 'git init' or try faking it with 'mkdir .git'.

jwangtobeno1, 2023-09-05:
only `touch .vim` also wokrs

JBlocklove, 2024-03-14:
According to the documentation it should fall back to the current working directory, which doesn't seem to be happening. I've encountered the same issue and while running `git init` solved it, that doesn't seem like it should be necessary.