chipsalliance/verible #1343: verible-verilog-ls is NOT working in Windows

r2com, 2022-06-03:
I tried getting it to work with neovim - failed, then I tried making it to work with Sublime - failed.

Below for example is troubleshooting log output from Sublime after all the setup was made in preferences as per manual below:
`{
    "servers": {
        "verilog": {
            "command": ["verible-verilog-ls"],
            "root": "",
            "url": "https://github.com/chipsalliance/verible"
        },
        "systemverilog": {
            "command": ["verible-verilog-ls"],
            "root": "",
            "url": "https://github.com/chipsalliance/verible"
        }
    }
}`

And here is troubleshooting output:
`# Troubleshooting: verible-verilog-ls
## Version
 - LSP: 1.16.2
 - Sublime Text: 4126
## Server Test Run
 - exit code: 0
 - output
```
Verible Alpha Language Server 0.0 alpha
```
## Server Configuration
 - command
```json
[
  "verible-verilog-ls"
]
```
 - shell command
```sh
verible-verilog-ls
```
 - selector
```
source.systemverilog
```
 - priority_selector
```
source.systemverilog
```
 - init_options
```json
{}
```
 - settings
```json
{}
```
 - env
```json
{}
```

## Active view
 - File name
```
C:\projects\temp\svtest.sv
```
 - Settings
```json
{
  "auto_complete_selector": "meta.tag, source - comment - string.quoted.double.block - string.quoted.single.block - string.unquoted.heredoc", 
  "lsp_active": null, 
  "syntax": "Packages/SystemVerilog/SystemVerilog.sublime-syntax"
}
```
 - base scope
```
source.systemverilog
```

## Project / Workspace
 - folders
```json
[]
```
 - is project: False

## LSP configuration

```json
{
  "clients": {
    "clangd": {
      "enabled": true
    }, 
    "verible-verilog-ls": {
      "command": [
        "verible-verilog-ls"
      ], 
      "enabled": true, 
      "selector": "source.systemverilog"
    }
  }
}
```
## System PATH

 - C:\soft\verible

 - `
After that, I went ahead and installed the whole thing in Debian Linux, and there I built it from source (in windows I just used binaries), after I built it from source I configured Sublime in a same way, and it could attach to ls and worked OK.

So this is now confirmed issue with Windows environment, please take a look!

p.s. I am not sure if it was ill-built binary or some other issue (I did not build binary for windows just used what you provided).

p.p.s: also, interesting fact is, in windows on bottom left, the status bar for verible-verilog-ls stuck with "Initialize" whereas in linux, where it works OK, it is there for 1 second and then just works.. 

so it seems its kinda starting but stuck somehow and does not function with client...

hzeller, 2022-08-30:
I need someone with access to a windows machine help figure out what is going on there.

hzeller, 2022-08-30:
I can suspect that windows implmentation of the `read()/write()` system calls might do some buffering, which they shouldn't do ?

hzeller, 2022-08-30:
Can you test the latest version ?

hzeller, 2022-09-09:
Did it work ?

AlphonseVA, 2022-09-26:
Hiya,

I am on **Windows 10** using **Sublime Text: Build 4126** and using LSP **v1.18.0**

I am on came to report and say yes it works.

Demo here:
[image]

Gutter works as expected.
Code actions work as expected.

The LSP: Toggle Diagnostics is working as expected.
[image]

The LSP: Toggle Diagnostics is working as expected.The LSP: Toggle Diagnostics is working as expected.
[image]:

hzeller, 2022-09-26:
Ah, wonderful, so it was apparently the non-binary-by-default output stream on windows that was then explicitly configured  in #1414 was the issue.
Thanks for confirming!

closed 2022-09-26 by hzeller