chipsalliance/verible #1363: No Highlight in VSCode

liuwuhaoo, 2022-07-15:
**Summary**

SystemVerilog code highlighting is not working in VSCode (Verible extension installed).

**Test cases and examples**

[image]

The file type is SystemVerilog.
The formater is working, the linter is working, but code highlight not.

**Additional context**
I downloaded released bin from [here](https://github.com/chipsalliance/verible/releases), Ubuntu-20.04-focal-x86_64, and copy bin files to /usr/bin.    
I built VS Code extension as the guideline says.

Thanks for everyone's help.

gasrodriguez, 2022-07-28:
You can try with another extension that provides highlighting, and wouldn’t interfere with verible ls. I use this [one](https://marketplace.visualstudio.com/items?itemName=gvekony.systemverilog-1800-2012).

jjlehner, 2022-11-07:
Any other extension I try which activates on System Verilog code seems to disable the Verible extension. Is this a way round this @gasrodriguez ?

> You can try with another extension that provides highlighting, and wouldn’t interfere with verible ls. I use this [one](https://marketplace.visualstudio.com/items?itemName=gvekony.systemverilog-1800-2012).