chipsalliance/verible #1381: Formatter crash found with fuzzing input; separator attachment heuristic issue ?

hzeller, 2022-08-11:
**Test case**

```c++
TEST(FormatterEndToEndTest, FuzzingRegressions) {
  FormatStyle style;
  std::ostringstream stream;
  absl::string_view input("P#(\0\0//\0//\0,);", 14);
  const auto status = FormatVerilog(input, "<filename>", style, stream);
  EXPECT_OK(status);
}
```

This results in some unhappy token_partition_tree

```
F0811 14:04:29.887721    2170 token_partition_tree.cc:64] Check failed: parent_end == children_end (0 vs. 6) node:
{ (>>>>[<auto>], policy: fit-else-expand) @{0,0,0}, (origin: "P#(
    //
...        ,
)")
  { (>>>>[P # (], policy: fit-else-expand, (origin: "P")) }
  { (>>>>>>>>[<auto>], policy: tabular-alignment) @{0,0,0,1}, (origin: ",")
    { (>>>>>>>>[//], policy: fit-else-expand) }
    { (>>>>>>>>[//], policy: fit-else-expand) }
  }
  { (>>>>[,], policy: uninitialized) }
}
```

Bisecting shows that this started to fail with the separator attachment heuristic https://github.com/chipsalliance/verible/commit/a45a3957e8e672d2f354c21b2baf47df182346a3

hzeller, 2022-08-11:
I've added a test to document

https://github.com/chipsalliance/verible/blob/101425ad37c4bd111f807cda9f63b148dba6f236/verilog/formatting/formatter_test.cc#L18218-L18225

mglb, 2022-08-22:
The problem in the mentioned comment is caused by assumption that if two tokens are in the same line (i.e. there's no `\n` between them) they can be joined: https://github.com/chipsalliance/verible/blob/master/verilog/formatting/tree_unwrapper.cc#L1536-L1545 .
However, the lexer/parser has different concept of "same line" - looks like `\0` does work as a line separator.
I see three ways of fixing this:
- treat `\0` as a line separator everywhere in the code,
- treat `\0` in the parser like a space (this would make second half of the example to be a comment) and print a warning,
- just abort parsing early due to `\0` in input stream (with nice error).

I would go with the second option if that's something doable in flex without tons of hacks. Otherwise the third option.

hzeller, 2022-08-26:
@fangism what do you think ?
I tend to think that NUL bytes in the input should be regarded invalid and considered a syntax error (option 3).