chipsalliance/verible #1397: Issue with IDE configuration.

karimtera, 2022-08-21:
Hello,

I have a probem with Neovim, Google test included headers are not found.
I am using coc-clangd as a extension for clangd in Neovim, and it works fine with `compile_commands.json` file.
However, I found that `*_test.cc` files are not found within `compile_commands.json` file, I tried to manually add an entry for a `_test.cc` file and it worked.

Is it a known limitation or an expected behavior?

mglb, 2022-08-22:
Hi.
It is a known limitation. Only targets [explicitly listed in `compilation_database` target](https://github.com/chipsalliance/verible/blob/7928006eab7892b9407cb543cc6bc5b9bfa00b6e/BUILD#L80-L92) are included in compile_commands.json.

As a workaround, you can use external script, `generate.py` (AKA `bazel-compdb`), from https://github.com/grailbio/bazel-compilation-database. It does include test sources when executed with `//...` target.

karimtera, 2022-08-22:
Thanks a lot!
I will try to use this one.

closed 2022-08-23 by mglb