chipsalliance/verible #1404: Installation not working on macOS after building the binaries

LegendaryPredz, 2022-08-24:
WARNING: Download from https://mirror.bazel.build/ftp.gnu.org/gnu/bison/bison-3.3.2.tar.xz failed: class java.io.FileNotFoundException GET returned 404 Not Found
WARNING: Download from https://mirror.bazel.build/github.com/jmillikin/rules_bison/releases/download/v0.1/bison-gnulib-788db09a9f88abbef73c97e8d7291c40455336d8.tar.xz failed: class java.io.FileNotFoundException GET returned 404 Not Found
WARNING: Download from https://mirror.bazel.build/ftp.gnu.org/gnu/m4/m4-1.4.18.tar.xz failed: class java.io.FileNotFoundException GET returned 404 Not Found
WARNING: Download from https://mirror.bazel.build/github.com/jmillikin/rules_m4/releases/download/v0.1/m4-gnulib-788db09a9f88abbef73c97e8d7291c40455336d8.tar.xz failed: class java.io.FileNotFoundException GET returned 404 Not Found
WARNING: Download from https://mirror.bazel.build/github.com/westes/flex/releases/download/v2.6.4/flex-2.6.4.tar.gz failed: class java.io.FileNotFoundException GET returned 404 Not Found
INFO: Analyzed target //:install (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //:install up-to-date:
  bazel-bin/_install_gen.sh
  bazel-bin/install
INFO: Elapsed time: 0.742s, Critical Path: 0.48s
INFO: 1 process: 1 internal.
INFO: Build completed successfully, 1 total action
INFO: Build completed successfully, 1 total action

It says that it was successful but there is nothing being installed. Any help?

hzeller, 2022-08-26:
How did you invoke bazel ?

LegendaryPredz, 2022-08-27:
> How did you invoke bazel ?

- Cloned the verible repo.
- Ran `bazel build -c opt //...` 
- Ran `bazel run -c opt :install -- ~/bin`
- Ran all tests (all succeeded) with `bazel test -c opt //...`

LegendaryPredz, 2022-08-27:
@hzeller 
- Cloned the verible repo.
- Ran `bazel build -c opt //...` 
- Ran `bazel run -c opt :install -- ~/bin`
- Ran all tests (all succeeded) with `bazel test -c opt //...`

hzeller, 2022-08-28:
And then the binaries did not show up in `~/bin` or `/usr/local/bin` (your two installation locations), so that it is not working ?

LegendaryPredz, 2022-08-29:
> And then the binaries did not show up in `~/bin` or `/usr/local/bin` (your two installation locations), so that it is not working ?

Correct. No binaries in either installation location.

LegendaryPredz, 2022-08-29:
Correct. There are no binaries showing in either installation location

hzeller, 2022-08-30:
the install routine should've been more resilient after https://github.com/chipsalliance/verible/pull/1396 because it uses the short options for the `install` program needed in BSD installations (such as macOS). Since you filed this after that was merged it sounds like there might be another reason why this is not working yet.

The strange thing is that there is no error message when you run install.
Question: does the directory `~/bin` or `/usr/local/bin` pre-exist on your machine ? It could be that `install` does not create the directories on your machine and just creates a file with that name containing the binary...

LegendaryPredz, 2022-08-31:
The directories `~/bin` and `~/usr/local/bin` already exist on my machine.

Yaotian-Liu, 2022-09-04:
I have a similar problem on m1 MacBook Air as well. Build and test works fine, but installation do not work. No related file is added to `~/bin` or `/usr/local/bin`.
Below is the output:

-> bazel run -c opt :install -- ~/bin
WARNING: Ignoring JAVA_HOME, because it must point to a JDK, not a JRE.
WARNING: Ignoring JAVA_HOME, because it must point to a JDK, not a JRE.
INFO: Analyzed target //:install (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
INFO: From BazelWorkspaceStatusAction stable-status.txt:
fatal: No annotated tags can describe '694974a68f104ef5d13d1a105d5e29f5a5f2db61'.
However, there were unannotated tags: try --tags.
Target //:install up-to-date:
  bazel-bin/_install_gen.sh
  bazel-bin/install
INFO: Elapsed time: 0.173s, Critical Path: 0.04s
INFO: 1 process: 1 internal.
INFO: Build completed successfully, 1 total action
INFO: Build completed successfully, 1 total action

tymcauley, 2023-01-23:
Ya, I just tried this myself. The problem happens at the very start of the `install` script:

```
# --- begin runfiles.bash initialization v2 ---
# Copy-pasted from the Bazel Bash runfiles library v2.
f=bazel_tools/tools/bash/runfiles/runfiles.bash
source "${RUNFILES_DIR:-/dev/null}/$f" 2>/dev/null || \
  source "$(grep -sm1 "^$f " "${RUNFILES_MANIFEST_FILE:-/dev/null}" | cut -f2- -d' ')" 2>/dev/null || \
  source "$0.runfiles/$f" 2>/dev/null || \
  source "$(grep -sm1 "^$f " "$0.runfiles_manifest" | cut -f2- -d' ')" 2>/dev/null || \
  source "$(grep -sm1 "^$f " "$0.exe.runfiles_manifest" | cut -f2- -d' ')" 2>/dev/null || \
  { echo>&2 "ERROR: cannot find $f"; exit 1; }; f=;
# --- end runfiles.bash initialization v2 ---
```

If you remove all of the redirects to `/dev/null`, then you get this output when running the install script:

```
$ ./bazel-bin/install ~/.local/bin
./bazel-bin/install: line 35: /dev/null/bazel_tools/tools/bash/runfiles/runfiles.bash: Not a directory
$ echo $?
1
```

Line 35 corresponds to that very first call to `source`. Not sure why, but it looks like the exit-code-chaining isn't working.

Changing this to supply `RUNFILES_DIR` gets a bit further:

```
$ RUNFILES_DIR=./bazel-bin/install.runfiles ./bazel-bin/install ~/.local/bin
install: illegal option -- T
usage: install [-bCcpSsv] [-B suffix] [-f flags] [-g group] [-m mode]
               [-o owner] file1 file2
       install [-bCcpSsv] [-B suffix] [-f flags] [-g group] [-m mode]
               [-o owner] file1 ... fileN directory
       install -d [-v] [-g group] [-m mode] [-o owner] directory ...
```

This looks like an issue with using BSD `install` vs GNU `install`. If you edit `./bazel-bin/install` to use `ginstall` as the binary (the GNU binary, provided by the [homebrew formula `coreutils`](https://formulae.brew.sh/formula/coreutils#default)), then the script succeeds, and installs the binaries as desired.

hzeller, 2023-01-23:
Does it help to just remove the `-T` flag in the `./baze-bin/install` ? Ideally that way we then can keep using the system `install` instead of relying on `ginstall`.

This is issue https://github.com/google/bazel_rules_install/issues/31 that I filed with bazel_rules_install a while back. Maybe pinging that could help.

tymcauley, 2023-01-24:
Yup, removing the `-T` flag made it work. I still had to set that environment variable to make that `source ... || ...` chain work.

hzeller, 2023-02-01:
I've now automatically remove the `-T` in the install script.
I don't have a Mac to test; can  you try if now building and installing on MacOS works smoothly from latest git ?

tymcauley, 2023-02-02:
Thanks @hzeller! That resolved the `install: illegal option -- T` error. At this point, we now still have to deal with the `source` issue in `bazel-bin/install`. This comment shows how I worked around that issue: https://github.com/chipsalliance/verible/issues/1404#issuecomment-1400726752

However, this does mean that I can't just run `bazel run -c opt :install -- ~/.local/bin`, that command still errors out. I have to use that workaround, manually running the command `RUNFILES_DIR=./bazel-bin/install.runfiles ./bazel-bin/install ~/.local/bin`. The good news now is that running `RUNFILES_DIR=./bazel-bin/install.runfiles ./bazel-bin/install ~/.local/bin` succeeds, and installs all of the binaries.

hzeller, 2023-04-17:
Unfortunately, I don't have access to a Mac, so can't really iterate on a solution here that works here and elsewhere.
Ideally, this is something that would be fixed https://github.com/google/bazel_rules_install by someone who knows their way around the MacOS environment.

The [homebrew package of Verible](https://github.com/chipsalliance/homebrew-verible)  side-steps the installation by manually enumerating all the binaries to install:

https://github.com/chipsalliance/homebrew-verible/blob/46664b28ed8f0dfe2d709faf45691cf42d9628b6/Formula/verible.rb#L34-L44

zhutmost, 2023-07-18:
I have tried to install from homebrew, or compile from the code. But neither works (same error during compilation). If you can compile one on the CI, I will be glad to help test the executable.

zhutmost, 2023-07-19:
Hi @hzeller , thanks for you quick reply.
I recompiled today. It looks a bit different two days ago (still fail but other bugs). See below: 
[image]
I am not familiar with Bazel/C++ compilation, so I have no idea to fix it. :(

I mean that, can we provide a prebuilt binary for homebrew (maybe one for intel and one for apple silicon)? Current installation from brew (i.e. compiling from code with Bazel) often fails.

hzeller, 2023-07-19:
Interesting, it looks like the compilation picks up a header from the system (`/usr/local/include`) in addition to the actual header it should use (`external/com_google_boringssl/...`). I have a suspicion, addressed in #1970

hzeller, 2023-07-19:
can you `git pull` and try again ?

zhutmost, 2023-07-20:
> can you `git pull` and try again ?

Hi @hzeller . I have cloned the latest repo, and rebuilt it. Still fails but other new bugs.

```log
~/Workspace/linter/verible master ···········································································································  base 10:13:06
❯ bazel build --jobs=12 -c opt --//bazel:use_local_flex_bison //...
Starting local Bazel server and connecting to it...
INFO: Analyzed 1898 targets (142 packages loaded, 5966 targets configured).
INFO: Found 1898 targets...
INFO: From Linking external/com_google_absl/absl/types/libbad_optional_access.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/types/_objs/bad_optional_access/bad_optional_access.o has no symbols
warning: /Library/Developer/CommandLineTools/usr/bin/libtool: archive library: bazel-out/darwin-opt/bin/external/com_google_absl/absl/types/libbad_optional_access.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking external/com_google_absl/absl/debugging/libdebugging_internal.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/debugging/_objs/debugging_internal/elf_mem_image.o has no symbols
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/debugging/_objs/debugging_internal/vdso_support.o has no symbols
INFO: From Linking external/com_google_absl/absl/log/liblog_entry.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/log/_objs/log_entry/log_entry.o has no symbols
warning: /Library/Developer/CommandLineTools/usr/bin/libtool: archive library: bazel-out/darwin-opt/bin/external/com_google_absl/absl/log/liblog_entry.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking external/com_google_absl/absl/base/libbase.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/base/_objs/base/unscaledcycleclock.o has no symbols
INFO: From Linking external/com_google_absl/absl/types/libbad_any_cast_impl.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/types/_objs/bad_any_cast_impl/bad_any_cast.o has no symbols
warning: /Library/Developer/CommandLineTools/usr/bin/libtool: archive library: bazel-out/darwin-opt/bin/external/com_google_absl/absl/types/libbad_any_cast_impl.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking external/com_google_absl/absl/types/libbad_variant_access.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/types/_objs/bad_variant_access/bad_variant_access.o has no symbols
warning: /Library/Developer/CommandLineTools/usr/bin/libtool: archive library: bazel-out/darwin-opt/bin/external/com_google_absl/absl/types/libbad_variant_access.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking external/com_google_absl/absl/crc/libcrc32c.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/crc/_objs/crc32c/crc_memcpy_x86_64.o has no symbols
INFO: From Linking external/com_google_absl/absl/strings/libcord.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/strings/_objs/cord/cord_buffer.o has no symbols
INFO: From Linking external/com_google_absl/absl/flags/libflag.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/flags/_objs/flag/flag.o has no symbols
warning: /Library/Developer/CommandLineTools/usr/bin/libtool: archive library: bazel-out/darwin-opt/bin/external/com_google_absl/absl/flags/libflag.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking external/com_google_absl/absl/types/libbad_variant_access.a [for tool]:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt-exec-2B5CBBC6/bin/external/com_google_absl/absl/types/_objs/bad_variant_access/bad_variant_access.o has no symbols
warning: /Library/Developer/CommandLineTools/usr/bin/libtool: archive library: bazel-out/darwin-opt-exec-2B5CBBC6/bin/external/com_google_absl/absl/types/libbad_variant_access.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking external/com_google_absl/absl/base/libbase.a [for tool]:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt-exec-2B5CBBC6/bin/external/com_google_absl/absl/base/_objs/base/unscaledcycleclock.o has no symbols
INFO: From Linking external/com_google_absl/absl/crc/libcrc32c.a [for tool]:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt-exec-2B5CBBC6/bin/external/com_google_absl/absl/crc/_objs/crc32c/crc_memcpy_x86_64.o has no symbols
INFO: From Linking external/com_google_absl/absl/strings/libcord.a [for tool]:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt-exec-2B5CBBC6/bin/external/com_google_absl/absl/strings/_objs/cord/cord_buffer.o has no symbols
[... cut ...]
Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
/Library/Developer/CommandLineTools/usr/bin/bison: unrecognized option `--report-file=bazel-out/darwin-opt-exec-2B5CBBC6/bin/verilog/parser/verilog.output'
Try `/Library/Developer/CommandLineTools/usr/bin/bison --help' for more information.
INFO: Elapsed time: 166.806s, Critical Path: 25.50s
INFO: 1416 processes: 699 internal, 717 darwin-sandbox.
FAILED: Build did NOT complete successfully
```

zhutmost, 2023-07-20:
[image]

zhutmost, 2023-07-20:
Note that I used local bison & flex installed by brew.

hzeller, 2023-07-20:
Can you tell me the version number of flex and bison on your machine ?
```
flex --version
bison --version
```

zhutmost, 2023-07-21:
OS:
Darwin XXXX-MacBookPro.local 22.5.0 Darwin Kernel Version 22.5.0: Thu Jun  8 22:22:22 PDT 2023; root:xnu-8796.121.3~7/RELEASE_X86_64 x86_64

bison: 3.8.2
flex: 2.6.4
I just downloaded them form homebrew and then added to PATH.

I have downloaded the prebuilt binary from the GITHUB Release page. I tried the verible-verlog-ls, and it looks good.

hzeller, 2023-07-21:
Ah good point. Now that I have added the macOS binaries in the release it should be possible to modify 

[`verilog/tools/ls/vscode/src/download-ls.ts`](https://github.com/chipsalliance/verible/blob/master/verilog/tools/ls/vscode/src/download-ls.ts) to not write the message anymore for macos, but downloads the published version instead.

Then [build it locally](https://github.com/chipsalliance/verible/tree/master/verilog/tools/ls#build-yourself) to test it it now downloads.

Since I don't have a mac to test, would you like to get this going and then send a pull request ?

zhutmost, 2023-07-21:
So are these messages just ignorable warning messages?!
I have added binary to PATH, but vscode do not use them in PATH. 
Looks that the function `checkIfBinaryExists(binaryPath: string)` doesn't work on macOS?

hzeller, 2023-07-21:
Yes, if the binary is in your path, it would just work.
If it does not find verible in the `$PATH`, it attempts to download it, so that is when the message is happening.

zhutmost, 2023-07-21:
ignore me. When I change the extension settings from the default “verible-verilog-ls” to a specific path “/...../...../...../bin/verible-verilog-ls”. it can use my downloaded one automatically.
Let me try your proposal. but tomorrow, it is too late today. :D

zhutmost, 2023-07-21:
> Yes, if the binary is in your path, it would just work.
> If it does not find verible in the `$PATH`, it attempts to download it, so that is when the message is happening.

Ahh, but I have added them to PATH. and it cannot find it, until I tell vscode its complete path.
command -v works well. 
[image]

zhutmost, 2023-07-21:
I am not sure whether it is just due to my local environment. So you can ignore it temporarily, if nobody else reports.

zhutmost, 2023-07-22:
@hzeller The situation is strange. I cloned the variable repo (master branch), and run bazel build:
```shell
bazel build --jobs=12 -c opt //...
```
and it failed. (still the multi header definition bug:
```log
~/Workspace/linter/verible master ························································ 6s  base 01:01:36
❯ bazel build --jobs=12 -c opt //...
Starting local Bazel server and connecting to it...
INFO: Analyzed 1898 targets (146 packages loaded, 6269 targets configured).
INFO: Found 1898 targets...
INFO: From Linking external/com_google_absl/absl/base/libbase.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/base/_objs/base/unscaledcycleclock.o has no symbols
INFO: From Linking external/com_google_absl/absl/types/libbad_optional_access.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/types/_objs/bad_optional_access/bad_optional_access.o has no symbols
warning: /Library/Developer/CommandLineTools/usr/bin/libtool: archive library: bazel-out/darwin-opt/bin/external/com_google_absl/absl/types/libbad_optional_access.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking external/com_google_absl/absl/flags/libflag.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/flags/_objs/flag/flag.o has no symbols
warning: /Library/Developer/CommandLineTools/usr/bin/libtool: archive library: bazel-out/darwin-opt/bin/external/com_google_absl/absl/flags/libflag.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking external/com_google_absl/absl/debugging/libdebugging_internal.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/debugging/_objs/debugging_internal/elf_mem_image.o has no symbols
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/debugging/_objs/debugging_internal/vdso_support.o has no symbols
INFO: From Linking external/com_google_absl/absl/crc/libcrc32c.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/crc/_objs/crc32c/crc_memcpy_x86_64.o has no symbols
INFO: From Linking external/com_google_absl/absl/types/libbad_any_cast_impl.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/types/_objs/bad_any_cast_impl/bad_any_cast.o has no symbols
warning: /Library/Developer/CommandLineTools/usr/bin/libtool: archive library: bazel-out/darwin-opt/bin/external/com_google_absl/absl/types/libbad_any_cast_impl.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking external/com_google_absl/absl/types/libbad_variant_access.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/types/_objs/bad_variant_access/bad_variant_access.o has no symbols
warning: /Library/Developer/CommandLineTools/usr/bin/libtool: archive library: bazel-out/darwin-opt/bin/external/com_google_absl/absl/types/libbad_variant_access.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking external/com_google_absl/absl/log/liblog_entry.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/log/_objs/log_entry/log_entry.o has no symbols
warning: /Library/Developer/CommandLineTools/usr/bin/libtool: archive library: bazel-out/darwin-opt/bin/external/com_google_absl/absl/log/liblog_entry.a the table of contents is empty (no object file members in the library define global symbols)
INFO: From Linking external/com_google_absl/absl/strings/libcord.a:
/Library/Developer/CommandLineTools/usr/bin/libtool: file: bazel-out/darwin-opt/bin/external/com_google_absl/absl/strings/_objs/cord/cord_buffer.o has no symbols
INFO: From Compiling gnulib/lib/vasnprintf.c [for tool]:
external/m4_v1.4.18/gnulib/lib/vasnprintf.c:3185:35: warning: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
                                  sprintf ((char *) p, "%+d", exponent);
                                  ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/stdio.h:188:1: note: 'sprintf' has been explicitly marked deprecated here
__deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")
^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
external/m4_v1.4.18/gnulib/lib/vasnprintf.c:3193:35: warning: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
[... cut ...] ill:/Users/zhutmost/Software/svls:/Users/zhutmost/Software/jetbrains-toolbox/scripts:/usr/local/Cellar/openjdk/20.0.1/libexec/openjdk.jdk/Contents/Home/bin:/Library/Java/JavaVirtualMachines/temurin-17.jdk/Contents/Home/bin:/Users/zhutmost/Library/pnpm:/usr/local/Caskroom/miniforge/base/bin:/usr/local/Caskroom/miniforge/base/condabin
```

I am not sure which part is different from your CI environment.
And I also forked your repo, but the CI cannot run on my account, due to the deployment setting.

zhutmost, 2023-07-22:
> I am not sure whether it is just due to my local environment. So you can ignore it temporarily, if nobody else reports.

The issue about that vscode cannot find lang server from PATH, is  fixed. It is due to that VScode won't update its PATH environment, until its process is completely killed and restarted. 

I still cannot compile from scratch.

hzeller, 2023-07-22:
Are you building from head ? The openssl issue should've been fixed with #1970

zhutmost, 2023-07-22:
@hzeller yes I am sure.

zhutmost, 2023-07-23:
Anyway we can change the homebrew rb setting to the prebuilt macOS binary now.

zhutmost, 2023-07-23:
It looks good on my Mac. 
Lint on the fly:
[image]
Format also works.

closed 2023-07-25 by hzeller