chipsalliance/verible #1444: Provide OpenSUSE binaries

hzeller, 2022-09-12:
OpenSUSE is a linux distribution used in the wild, so it would be good if we could provide release binaries for it as well.

(context: https://groups.google.com/g/verible-users/c/Z-iTXOakQUY )

hzeller, 2022-09-12:
Unai, do you think this would be easy to do ?

hzeller, 2022-09-13:
The user in question uses SUSE Enterprise Server 12.

umarcor, 2022-09-14:
I can't find SUSE Enterprise Server containers. There are https://hub.docker.com/r/opensuse/tumbleweed and https://hub.docker.com/r/opensuse/leap. Hence, it should be possible to generate pre-built binaries for opensuse, but I doubt those can be guaranteed to work on different versions of it.

See also:

- https://github.com/f4pga/ideas/issues/59
- https://build.opensuse.org/project/show/hardware:FPGA
- https://build.opensuse.org/package/show/home:seanmao/verible

hzeller, 2022-09-14:
Might be already a good starting point. In particular if we bulid the binaries mostly static s we already do for the rest ( `BAZEL_LINKOPTS` / `BAZEL_LINKLIBS`).

(In the particular case, it turned out that the user was then abel to run binaries we built for Ubuntu 16, so that particular distribution might've been pretty old, libc-wise)

hzeller, 2022-09-14:
While at it: RHEL is also often used in some enterprise contexts. Are there also RHEL containers we can use to build for these ?

umarcor, 2022-09-14:
With regard to RHEL, see https://github.com/chipsalliance/verible/issues/1301.

hzeller, 2023-04-14:
Since we now have static binaries https://github.com/chipsalliance/verible/releases
... more efforts for various Linux distributions might not be needed anymore