chipsalliance/verible #1467: Same target built multiple times ? Potential slowing down build.

hzeller, 2022-09-19:
Building shows some characteristic warnings that show up multiple times and seem to indicate that the build rule does not entirely keep track of the dependency (see at the bottom for the log).

The verilog lex warnings are emitted twice, and the warnings from verilog tab seem to be emitted three times.

On interesting observation is, that the generated verilog.tab.cc show up in two different directories, and they are slightly different in size:

```
$ ls -l bazel-out/k8-opt/bin/verilog/parser/verilog.tab.cc bazel-out/host/bin/verilog/parser/verilog.tab.cc
-r-xr-xr-x 1 user users 1457311 Sep 18 22:36 bazel-out/host/bin/verilog/parser/verilog.tab.cc
-r-xr-xr-x 1 user users 1461273 Sep 18 22:36 bazel-out/k8-opt/bin/verilog/parser/verilog.tab.cc
```

The size difference can be explained with two different `#line` emits:
```diff
26023c26023
< #line 26024 "bazel-out/host/bin/verilog/parser/verilog.tab.cc"
---
> #line 26024 "bazel-out/k8-opt/bin/verilog/parser/verilog.tab.cc"
```

Maybe the way the bzl rules for the flex bison invocation are not waterthight and get bazel confused if it needs to re-run them ? Maybe it gets confused because running the same code generation in different output directories ?

In this example, local flex and bison is used to reduce the chatter of warnings when compiling the hermetic flex and bison versions from the WORKSPACE. But same thing probably also happens without `--//bazel:use_local_flex_bison`, but in that case bison does not emit a warning so we only see the multiple instances of the lexer warning.

```
$ bison --version
bison (GNU Bison) 3.8.2
$ flex --version
flex 2.6.4
$ bazel --version
bazel 4.2.2- (@non-git)
```

```
$ bazel build -c opt --//bazel:use_local_flex_bison //...
INFO: Analyzed 1880 targets (125 packages loaded, 5835 targets configured).
INFO: Found 1880 targets...
INFO: From Executing genrule //verilog/parser:verilog_lex:
verilog/parser/verilog.lex:299: warning, rule cannot be matched
verilog/parser/verilog.lex:950: warning, rule cannot be matched
INFO: From Executing genrule //verilog/parser:verilog_lex [for host]:
verilog/parser/verilog.lex:299: warning, rule cannot be matched
verilog/parser/verilog.lex:950: warning, rule cannot be matched
INFO: From Executing genrule //common/analysis:command_file_lex [for host]:
common/analysis/command_file.lex:77: warning, rule cannot be matched
INFO: From Compiling verilog/parser/verilog-final.tab.cc:
bazel-out/k8-opt/bin/verilog/parser/verilog.tab.cc: In function 'int verilog::verilog_parse(verible::ParserParam*)':
bazel-out/k8-opt/bin/verilog/parser/verilog.tab.cc:26198:1: warning: label 'yyexhaustedlab' defined but not used [-Wunused-label]
INFO: From Compiling verilog/parser/verilog-final.tab.cc [for host]:
bazel-out/host/bin/verilog/parser/verilog.tab.cc: In function 'int verilog::verilog_parse(verible::ParserParam*)':
bazel-out/host/bin/verilog/parser/verilog.tab.cc:26198:1: warning: label 'yyexhaustedlab' defined but not used [-Wunused-label]
INFO: From Executing genrule //common/analysis:command_file_lex:
common/analysis/command_file.lex:77: warning, rule cannot be matched
INFO: From Compiling verilog/parser/verilog-final.tab.cc:
bazel-out/k8-opt/bin/verilog/parser/verilog.tab.cc: In function 'int verilog::verilog_parse(verible::ParserParam*)':
bazel-out/k8-opt/bin/verilog/parser/verilog.tab.cc:26198:1: warning: label 'yyexhaustedlab' defined but not used [-Wunused-label]
INFO: Elapsed time: 219.923s, Critical Path: 218.06s
INFO: 6737 processes: 3374 internal, 3363 linux-sandbox.
INFO: Build completed successfully, 6737 total actions
```

hzeller, 2022-09-19:
@QuantamHD what are good ways to debug such things with bazel ?

QuantamHD, 2022-09-19:
You can use aquery to find out which two actions are generating these files as a first suggestion