chipsalliance/verible #1475: Last CentOS 6 build

gogito, 2022-09-20:
Hi, can someone help point me to the final CentOS 6 build? My company is still using CentOS 6 so I can't use the newer builds.

Thanks in advance!

hzeller, 2022-09-20:
Dont' remember, but it probably started failing. Cent OS 6 is not updated since 2017 and [was EOL November 2020](https://wiki.centos.org/FAQ/General#What_is_the_support_.27.27end_of_life.27.27_for_each_CentOS_release.3F) and at that point often things fall apart due to unmaintained/removed containers and packages (You probably experience that as well since you have to use Cent OS 6). So probably around May it started failing so it was finally removed in July.

If you can get the release script to work to build Cent OS 6 releases, we're happy to accept pull requests.

gogito, 2022-09-21:
You're right that it's so old that it's a pain to compile anything that can work on it, but I've worked for multiple hardware company and most of them are still on Cent/RHEL 6 which is unfortunate. I will try to find a way to compile it and report back. It would be nice if we still have the centOS 6 branch so I can get a starting point.

hzeller, 2022-09-21:
This was the patch that removed it ... so maybe start with the reversse patch
https://github.com/chipsalliance/verible/commit/3b9a0750398a29398c4dbd8981a2fbb457d6cd64.patch

Also paging @umarcor who can give more context. I suspect it was removed due to age.

hzeller, 2023-04-14:
We now also distribute statically linked binaries in the releases:
   https://github.com/chipsalliance/verible/releases
Can you try if that would work on your system ?