chipsalliance/verible #1498: syntax error in macro processing

matutem, 2022-10-07:
Verible fails to process valid macro.

When processing a macro that contains a comment verible gets confused. The comment uses /``/ so it ends up with // after processing it. This `` expansion is allowed by the LRM. VCS accepts this with no problem..

**To Reproduce**

```systemverilog
`define PRAGMA_COVERAGE_ON  \
  /``/VCS coverage on

`define PRAGMA_COVERAGE_OFF \
  /``/VCS coverage off

module test (input logic in);

  always_comb begin : src_fsm
    unique case (in)
  `PRAGMA_COVERAGE_OFF
      default: ;
  `PRAGMA_COVERAGE_ON
    endcase
  end
endmodule
```

Include any options used.

**Actual behavior:**

verible gets a syntax error.

**Expected behavior**

The expectation is that the PRAGMA_COVERAGE_ON macro will expand to the following:
//VCS coverage on

The reason is that when the text is parsed it will not be considered a straight comment because of the /``/ token, but after expansion the two slashes will be joined per IEEE 1800-2017 section 22.5.1, page 680, and the expansion will be a comment.

hzeller, 2022-10-11:
which of the verible tools were you using ? The verible-verilog-syntax for instance does not do preprocessing yet (work in progress to wire that up), but the standalone preprocessor does so it might work with that.

Adding @karimtera to verify if the preprocessor can deal with it.

karimtera, 2022-10-11:
The preprocessor doesn't handle the concatenation double back-ticks so far.
I don't remember if we have a TODO for this or not, let me check.

matutem, 2022-10-24:
I was using verible-verilog-format.

sriyerg, 2022-12-07:
Is there an update on this issue? We would like to ramp up the usage of this type of macro in our codebase.