chipsalliance/verible #1502: Separate ownership of file content from analysis results

hzeller, 2022-10-11:
Currently, we have [`TextStructure`](https://github.com/chipsalliance/verible/blob/42782d267f11a9a754a774f6f3426bf04783bc0c/common/text/text_structure.h#L235-L268), which contains the content of the file and a `TextStructureView` (which essentially holds parse results on top of that).
Then `FileAnalyzer` [inherits from TextStructure](https://github.com/chipsalliance/verible/blob/42782d267f11a9a754a774f6f3426bf04783bc0c/common/analysis/file_analyzer.h#L87) to generate specific analysis results.

This is somewhat rigid, as it means that any ownership of file content within Verible needs to be traced back to some `FileAnalyzer` or at least `TextStructure`.

This is a bit rigid. It couples the ownership of the file content memory tightly with the analyzers but that is not necessarily desirable.

These days, we have more requirements that warrants separating opening files and keeping their content in memory from the analysis steps working on them.
  * There might be multiple analyzers corresponding to a file possibly with multiple configuration options. Right now, this requires multiple copies of the same content.
  * Sometimes, for sub-analysis, we have to copy file contents (as they are rigidly associated with analyzers), instead of just carving out string_views.
  * A preprocessed file might need to open multiple files and then create a uniform token stream on top of that. The same include file might be needed in multiple files. That means there is no 1:1 correspondence anymore between file contents and various views referring to it.
  * The Kythe indexer also possibly opens multiple files with includes, but mostly works file-by-file. Given that we want to index these sequentially (we emit facts as soon as we discover them), we also might need to retire files from memory once not needed anymore (i.e. referenced by any string view in memory). But slightly prefer files that were included to keep in memory longer as we might see them included in later files as well.
  * Other applications, such as the formatter, only need a subset of the analysis (token stream, but not necessarily full parsing), and are strictly work file-by-file and always want to purge files once done.
  * Once file content ownership is nicely stored in [`MemBlock`](https://github.com/chipsalliance/verible/blob/master/common/strings/mem-block.h#L27-L40)s, these can even come from a `mmap()` if supported by operating system and filesystem.
  * _Possibly more requirements; add them to this issue._
  
This issue is the anchor for refacoring the current situation to meet these goals. Possible path:
  * Adapt `VerilogProject` to be the owner of file contents of which 'views' (token stream, concrete syntax tree) then can work with. Keep track of multiple lifetimes (files to be purged, files to be maybe purged but good to keep around if possible, such as include files that might be needed later). This will also be the place, given a string-view, can determine the file-name and line number.
  * Have separate views on that data that can be used and composed (currently everything in `TextStructureView`).
    * There can be a token stream of one file, or multiple files as a result of preprocessing. 
    * Then a syntax tree can be parsed from that. 
    * The VerilogProject and its files keeps track of line-numbers, otherwise not needed except for emitting error messages etc. 
    * The decomposition makes it easy to have each tool only generate what it needs. Also it is not necessary to have awkward state machines (like in `VerilogFile` right now), as there can be different objects representing different parts. A parse view naturally requires that the token stream view is ready first etc.
  * Ownership needs to be decided if we just have central `unique_ptr<MemBlock>` maybe require shared_ptr<> to hand out.
  
So far done
  * #1491
  * #1492
  * #1493

**This is a collecting issue for ideas and PRs submitted towards the goal**

hzeller, 2022-10-11:
CC: @fangism @ivan444 @mglb @karimtera for comments.

hzeller, 2022-10-11:
> Who'll keep the ownership of the analysis data when it's not pure derivative of the input files

Good question, I left that out for simplicity at this point. I see a two-step approach
  1. Initially, just do this as it is now and StrCat() these and store as new string with a constructred name.
  2. Later, compose this as if this was an include and just concatenate the token stream. The strings then are stored in the  'filename -> stringcontent' container and have names such as `<module-prefix>` in case they ever show up in error messages. The exact implementation needs to be determined, but probably extending the idea of a 'file content' as an array of string views could be considered.

Speaking of file name mappings: also the file-list and command-line will just be stored as named string sources. That way we can have messages that refer to e.g. `+define+` macros in these locations.  So hypothetically:

```
foo.sv:3:17: Redefinition of macro `FOOBAR
<commandline>:0:42: first defined here.
```

fangism, 2022-10-11:
I'm partial to having separate-and-composable views of text and data.  This way every analysis will operate on a need-to-know basis.  `VerilogProject` is but one abstraction for doing useful work, but there can be other lighter-weight ones.  (As mentioned, formatting needs very little.)