chipsalliance/verible #1524: Mismatched token enum in formatter output: ')' vs. MacroCallCloseToEndLine

anjanaskumar27, 2022-10-20:
**Test case**

```systemverilog
module a (
  input wire [`clog2(2):0] b
);
endmodule
```

**Actual output**

```systemverilog
// This doesn't look right.
module a (
    input wire [
    `clog2(2)
    :0] b
);
endmodule
```

Include any possible diagnostic messages from the formatter.

**Expected or suggested output**

```systemverilog
Formatted output is lexically different from the input.    Please file a bug.  Details:
Mismatched token enums.  got: (')') (#41: ")") vs. (MacroCallCloseToEndLine) (#711: ")")
First mismatched token [1974]: (')') (#41: ")") vs. (MacroCallCloseToEndLine) (#711: ")")
```

Citations to published style guides would help.

hzeller, 2022-12-07:
Looks like a closing parenthesis of a macro call at the end of a line gets its own token `MacroCallCloseToEndLine`, possibly to be a useful distinguishing factor in the [uvm semicolon rule](https://github.com/chipsalliance/verible/blob/39ed2e3f737b8f8bc61dc5fbe99be09091f63e40/verilog/analysis/checkers/uvm_macro_semicolon_rule.cc#L89).

Two things come to mind here
  * Maybe for comparison of token enums we should (probably only a few ?) equivalence classes. That way the triggering [enum comparison](https://github.com/chipsalliance/verible/blob/39ed2e3f737b8f8bc61dc5fbe99be09091f63e40/verilog/analysis/verilog_equivalence.cc#L170) could see this as equivalent.
  * Independent of that: Fix the formatter to actually not add a newline after a macro in these circumstances ``wire [`foo():0]``

hzeller, 2023-04-20:
b/230396830

hzeller, 2023-04-25:
Since you're already working towards a solution, assigning to you @jbylicki

hzeller, 2023-05-05:
With the recent changes, it at least does not complain anymore, but the formatting result is ... not super pretty, indicating that some of these tokens are not well classified in the formatter

```systemverilog
module a (
    input wire [
    `clog2(2)
    :0] b
);
endmodule
```

kbrunham-intel, 2026-09-22:
@hzeller I retested the reduced case from this issue with `verible-verilog-format` built from tag `v0.0-4294-gc1d8f5e8`. I could not reproduce the problem (formatter exits 0; output keeps the expected layout / converges). I would suggest closing this issue.

`input wire [\`clog2(2):0]` formats on one packed-dimension line without token-enum mismatch.