chipsalliance/verible #1573: Smoke tests try to validate files that should not be ran in single-file mode

jbylicki, 2022-12-29:
The problem relates to a number of files in the smoke-tests, which are taken from the sources and put into the `-syntax` and `-lint` tools without any context or processing. Some files, especially `.svh` are written such that they should be embedded via `include` to other files and should not be parsed on their own.

An example would be an `opentitan` case, where `hw/dv/sv/cip_lib/seq_lib/cip_base_vseq__sec_cm_fi.svh` is throwing a `rc=1` return code because of the 26 syntax errors that are produced while parsing this file on its own. The main issue in this particular case is that `virtual task`s declared in that file are there on their own, however the parser expects those keywords to appear only within a class declaration. This file is included into `hw/dv/sv/cip_lib/seq_lib/cip_base_vseq.sv` inside of the `cip_base_vseq` class and within this context it is valid.

I believe that we should somehow limit what files are getting into the smoke tests, such that unnecessary errors like those can be avoided.

kgugala, 2022-12-29:
CC @hzeller

hzeller, 2023-01-07:
Yes, currently it just takes all files that somehow looks like they contain Verilog.
In particular the project tool is of course not very happy with single files.

If there are single, incomplete files (such as the *.svh mentioned above), we could possibly manually exclude them. But I suspect it would be more beneficial to create some reasonablish file-list and then process that.

From looking at it, what do you think would be a good first step @jbylicki ?

jbylicki, 2023-02-01:
The smoke test analysis script looks whether there are syntax errors in `.svh` files and if so, labels them as files that should not be taken out of context. I would propose as a reasonable start to extract those via the analysis script (manually) and implement a list of skipped files in the `smoke-test`. For now said list would contain only those `.svh`-es, but it could be extended down the line if more files turn out problematic in that way. Is this a good approach?